JelmerT/cc2538-bsl #16: Reading memory results in exception

svenKautlenbach, 2015-02-18:
When trying to read memory I get "ERROR: ord() expected string of length 1, but int found".
When removing the mdebug at the read section, I get similar error when it tries to write data to file - "ERROR: sequence item 0: expected string, int found".

svenKautlenbach, 2015-02-18:
Ok, I think I managed to get it correct. Maybe it is the fault of python 2.7?
mdebug(5, " 0x%x: 0x%02x%02x%02x%02x" % (conf['address']+(i*4), rdata[3], rdata[2], rdata[1], rdata[0]), '\r')
file(args[0], 'ab').write(''.join(str(reversed(rdata))))

JelmerT, 2015-02-19:
Interesting bug, thanks for reporting it.

Are you getting back the correct data after making your proposed fix? It sounds to me that `cmdMemRead` is returning something that it shouldn't into `rdata`

svenKautlenbach, 2015-03-13:
Hi!

Sorry for slow response.
Actually this fix did not work, it just enabled to run the script with no problems, but inside the memory file was still incorrect data. And I have lost the file now, but there was text something about the iterator or object at 0x######. Can you assert which python version you are specifically running.

JelmerT, 2015-03-16:
The script should work with both 2.7 and 3.4. 
Are you maybe willing to share the firmware file with me, so I can try to reproduce the error? If you still have it.

svenKautlenbach, 2015-03-16:
It is Contiki build, I simply am not allowed to share the specific binaries :(

JelmerT, 2015-03-20:
I understand, can you share the command you used to try to read the memory? Could have something to do with the length and location.

svenKautlenbach, 2015-03-30:
Sorry all gone by now. I will try to reproduce it when I have time, I will just try out and experiment with the script. Could you please hold on for now?

JelmerT, 2015-04-17:
No problem, I'll close this for now, feel free to comment when you come across the problem again.

closed 2015-04-17 by JelmerT