JelmerT/cc2538-bsl #17: Errno 22 Invalid argument

svenKautlenbach, 2015-03-30:
From time to time with no reason I get this. First I started to get it when I specified baud rate with "-b". Then removed the option and got back okay again. Then later I started to get this with "-p" option. And then I started to test what exactly is wrong.
18:42:43-user@instank-contriki:~/Folder/ ((0a8645b...))$ python cc2538-bsl.py
ERROR: could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
18:42:51-user@instant-contiki:~/Folder/ ((0a8645b...))$ sudo python cc2538-bsl.py
ERROR: [Errno 22] Invalid argument

Can you explain what is the cause of this? Seems like there is nothing to do with the arguments? More and more I start to feel like there is something wrong with the serial fd.

JelmerT, 2015-04-17:
The 'Invalid argument' probably has something to do with the previous error, where it can't open the serial port. Is maybe some other process using that port?

The '[Errno 22] Invalid argument' does indeed not refer to the arguments used when calling the script but rather a problem with a function in the script itself.

JelmerT, 2015-07-10:
Were you able to solve this @svenKautlenbach ?

svenKautlenbach, 2015-07-29:
Not sure 100%, but the hypothesis you suggested might have been correct that something else was using the serial port too. I had 6LBR running, that has caused problems for me before too...
Can close it.

closed 2015-08-07 by JelmerT