JelmerT/cc2538-bsl #25: Support for the CC26xx

g-oikonomou, 2015-05-23:
So the CC26xx has a pretty similar (I suspect identical) bootloader protocol as the CC2538. The main difference is that code starts at 0x00000000 instead of 0x200000, flash size is limited to 128KB, different chip identifier. I have configured my chip to enter bootloader mode

* The erase command seems to ignore the address argument and it would be nice to make size configurable, thus changes required here:

```
$ git diff
diff --git a/cc2538-bsl.py b/cc2538-bsl.py
index 32565b6..55ba7a7 100644
--- a/cc2538-bsl.py
+++ b/cc2538-bsl.py
@@ -718,8 +718,8 @@ if __name__ == "__main__":
 
         if conf['erase']:
             # we only do full erase for now (CC2538)
-            address = 0x00200000 #flash start addr for cc2538
-            size = 0x80000 #total flash size cc2538
+            address = conf['address'] # flash start addr for cc2538
+            size = conf['size'] # total flash size cc2538
             mdebug(5, "Erasing %s bytes starting at address 0x%x" % (size, address))
 
             if cmd.cmdEraseMemory(address, size):
```
* But even after fixing address and size, there still seems to be a problem - see below. I will try to track it down but pointers would be welcome. The fact it works fine with CC2538 suggests it could be related to the chip ID?

```
$ python ../../tools/cc2538-bsl/cc2538-bsl.py -a 0 -e -V -b 500000
Opening port /dev/tty.usbserial-00002014B, baud 500000
Connecting to target...
*** sending synch sequence
*** GetChipId command (0x28)
*** received 6 bytes
*** GetStatus command (0x23)
*** received 3 bytes
Command Successful
Warning: unrecognized chip ID 0xf000
Erasing 131072 bytes starting at address 0x0
*** Erase command(0x26)
*** GetStatus command (0x23)
*** received 3 bytes
ERROR: unhashable type: 'list'

```

* The write command seems to program the chip correctly.
* It looks like I'll need to look at verify too. Any pointers?:
```
$ python ../../tools/cc2538-bsl/cc2538-bsl.py -a 0x00000000 -b 500000 -V -v -w cc26xx-demo.bin 
Opening port /dev/tty.usbserial-00002014B, baud 500000
Reading data from cc26xx-demo.bin
Connecting to target...
*** sending synch sequence
*** GetChipId command (0x28)
*** received 6 bytes
*** GetStatus command (0x23)
*** received 3 bytes
Command Successful
[128, 2, 240, 0]
Warning: unrecognized chip ID 0xf000
Writing 131072 bytes starting at address 0x0
*** Download command (0x21)
*** GetStatus command (0x23)
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x0
*** GetStatus command (0x23)
*** received 3 bytes
Command Successful
*** Send Data (0x24)
*** GetStatus command (0x23)
*** received 3 bytes
Command Successful
    Write done                                
Verifying by comparing CRC32 calculations.
*** CRC32 command(0x27)
*** received 6 bytes
*** GetStatus command (0x23)
*** received 3 bytes
Command Successful
*** Reset command (0x25)
ERROR: NO CRC32 match: Local = 0x909cd098, Target = 0xffffffff
```

g-oikonomou, 2015-05-23:
Scrap the last one: Verify works fine it's just the protocol is slightly different.

JelmerT, 2015-05-26:
This looks very interesting! I looked for references to a bootloader in the datasheets before, but was pretty sad when I didn't find anything. But this is great news.

I have the chips, but no dev boards yet, once I have a working board I'll start testing.

g-oikonomou, 2015-05-26:
Basically the transport is identical, but there are the following major differences in terms of commands:

* Erase is now handled by `COMMAND_BANK_ERASE` (0x2C and _not_ 0x2A which is what the CC26xx-TRM says). This erases the entire user area of the flash, so this is a simple 3-byte command (LEN, CRC, CMD), but does not need an address. So my comment regarding how erase ignores `-a` can be ignored itself!
* `COMMAND_CRC32` now takes an additional 3 bytes which specify the "Read Repeat Count". This can be a simple `0x00000000`.
* There is no `COMMAND_SET_XOSC` (none documented anyway), so we have to force -b 500000.
* The expected return value of `COMMAND_GET_CHIP_ID` is unclear to me.

The CC26xx bootloader is documented in section 8 of the CC26xx TRM (SWCU117A), but the command IDs for `COMMAND_BANK_ERASE`, `COMMAND_SET_CCFG`, `COMMAND_MEMORY_READ`, `COMMAND_MEMORY_WRITE` are incorrect (they correspond to an older chip version). I have flagged this with TI already. Unless one is using said early chip version, the correct command IDs are:

```
CMD_MEMORY_READ      = 0x2A,
CMD_MEMORY_WRITE     = 0x2B,
CMD_BANK_ERASE       = 0x2C,
CMD_SET_CCFG         = 0x2D,
```

With those changes in place, the script fundamentally works. I started preparing a pull, but I quickly found myself trying to re-engineer big parts of the script and I ran out of time. As far as I can tell though, that's about it in terms of changes required. Chip auto-detection should be trivial based on `GET_CHIP_ID`.

JelmerT, 2015-05-27:
Oh, it is document indeed, I probably looked too soon or looked over it.
Thanks for figuring out TI's errors in the datasheet!

I'll create a new branch for the implementation of CC26XX support and start adding your findings. Are there any particular parts of the script you think need re-engineering?

`COMMAND_GET_CHIP_ID` seems to return the `USER_ID` Register in which we can check bit 27-26 `VER` for 0x0. Which is probably not the best way to check if it's a CC26xx, but the only one available I think.

g-oikonomou, 2015-05-27:
> Oh, it is document indeed, I probably looked too soon or looked over it.

Probably too soon. If memory serves, the 1st online version didn't have this info.

> COMMAND_GET_CHIP_ID seems to return the USER_ID Register in which we can check bit 27-26 VER for 0x0. Which is probably not the best way to check if it's a CC26xx, but the only one available I think.

I think what I ended up doing was to assume CC26xx unless I got the CC2538 identifier.

> Are there any particular parts of the script you think need re-engineering?

I wouldn't go as far as say "need", it's just that this very big `CommandInterface` combines many logical functionalities and this makes it a little hard to extend. I was thinking about breaking things down:

* A class `Command` would have an initialiser with `command_id` and `data`, it would be able to calculate its own checksum. It could even be subclassed per different command to do length checking.
* A separate class `Transport` could be used to send commands, wait for ACK/NACK, send ACK/NACK over a `PHY` abstraction
* The `PHY` abstraction would now abstract PySerial in view of adding more `PHY` layers in the future, e.g. a FIFO for unit testing.
* A class `Device` would abstract operations such as erase, detect_chip, write an image, get/set IEEE address, read flash
* Lastly, in order to add support for a new device one would simply have to extend `Device`.

With those in place, "main" would simply do 1. Detect chip, 2. Set High Baud Rate (would simply not switch without erroring out for the CC26xx) 3. Do whatever the user asked for. So basically "main" would perform the exact same sequence of steps irrespective of chip.

I started working on all of the above at the same time and it ended up becoming a train wreck :)

JelmerT, 2015-06-06:
Added the CC26xx branch and just found this document: `CC2538/CC26xx Serial Bootloader Interface` http://www.ti.com/lit/an/swra466a/swra466a.pdf and the soft/source: http://www.ti.com/lit/zip/swra466
That should help a lot while implementing the CC26xx support. And it seems like the CC13xx has exactly the same bootloader, so lets test the branch against that chip too.

Seems like I'll need a new name for this script :/

Your proposed changes to the script make sense indeed. They'll make it more structured and more easily expandable. Lets try to implement them one by one (so it doesn't become a train wreck ;) ), but basic CC26xx support is the priority for now.

schuschu, 2015-06-08:
I am currently trying out the cc26xx branch on a sensortag with the cc-dev-backpack (backdoor and bootloader enabled, pin no 0, active low). Unfortunately I get no reply to the synch sequence. You wouldn't have tried this with a sensortag by any chance? (also I have rev 1.0.0 devpacks so I am sure the mixed up rx/tx will prevent me from ever using bsl, just checking)

JelmerT, 2015-06-08:
I just made the branch, didn't do any implementing yet. So it's normal that you're getting no replies yet. ;)

I'll start implementing based on what @g-oikonomou found and the datasheets, once I finish / receive my hardware. But feel free to start yourself and send a pull request!
Didn't know about the error in the 1.0.0 devpacks, but it indeed seems like you won't be able to properly access TX/RX without some hard core hardware hacking. I don't think there's a way to modify the bootloader to compensate for that routing error.

g-oikonomou, 2015-06-08:
According to the CC26xx TRM [1], the bootloader maps signals to pins statically. For QFN48 / 7x7 the mapping is:

```
UART_RX -> DIO_2
UART_TX -> DIO_3
```

This mapping is fixed and cannot be changed (The ability to change this in CCFG would have been nice, but as far as I can tell it is not possible).

This mapping corresponds to how the XDS UART backchannel connects to attached EMs. Since debugger devpacks physically connect RX and TX to DIO_28 / and _29 (swapped around between the two versions), I think the bootloader will never work with either version of the devpack :(

[1]  "CC26xx SimpleLink Wireless MCU Technical Reference Manual (Rev. A)" SWCU117A–February 2015–Revised March 2015, Table 8.3, section 8.2.2

schuschu, 2015-06-08:
I think this is more than just a devpack problem. I don't think DIO_2 and DIO_3 are accessible on the sensortag at all which means (especially when using contiki under linux) uploading will remain troublesome (with the smartrf tools only available under windows and uniflash ... well it sometimes happens to work but only for one node at one with no way of specifying). In your experience, is there any chance of using a second (software only) bootloader to achive this or cmsis-dap (which the devpack seems to support)?
I know this is somewhat of topic but I was really hoping for a bsl implementation similar to what was available for the telosb (sky) motes but that seems impossible now...

timothybrown, 2015-06-22:
For what its's worth, Energia has support for the XDS110 on the MSP432, which is also what's on the Debug DevPack. So, one should be able to use the XDS binary from Energia under OS X or Linux to flash the Sensortag.

In fact, I'll work on that today.

JelmerT, 2015-07-10:
Seems to me that there is a need for a better, simpeler dev board, with easier programming and sensor-hookup possibilities, targeted to the WSN-crowd. 
Any takers? :) It'll be hard to beat the 15$ price point of TI though.

@timothybrown any luck with those XDS110 drivers from Energia?
@schuschu a software only bootloader is definitely possible, you'll need to write it though ;) The telosb (and sky and z1) used the msp430-bsl script which is function-wise very similar to this script.

timothybrown, 2015-07-11:
@JelmerT I made progress, but under OS X I'm being held up by a single dynamic library!

Long story short, the MSP432 is a CM4, whereas the SensorTag is a CM3, so despite both boards using the same emulator a different set of config files are needed on the host. What I did was copy all the relevant files from CCS UniFlash Windows to OS X. The only thing missing is 'tixds510icepick_c.dylib' which I can't provide since I don't have access to TI's source code.

Now, if you're on Linux, you should be able to copy a working lib from CCS UniFlash Linux and get things working. In fact, I'll try this. 

TI is working on this for OS X, as it's currently preventing SensorTag debug from working with CCS Cloud on OS X. I've been poking at them for months about this. All they need is a single library to get things working on OS X! See: http://e2e.ti.com/support/wireless_connectivity/f/538/p/418011/1548976

JelmerT, 2015-07-29:
Thanks for the update @timothybrown, good work on pushing TI on this!

JelmerT, 2015-09-02:
Everyone: @g-oikonomou added a new branch to this repo with CC26XX support.

Has anyone used or tested this new implementation? Is everything working as expected? I'd love some feedback on it so we can decide how to continue and implement it properly into the master branch.

closed 2015-09-19 by JelmerT

JelmerT, 2018-11-01:
@arunpm04 Looks like there's no big changes to the bootloader for CC26x2, so my guess would be yes. But if you run into any problems, feel free to open an issue or pull request