JelmerT/cc2538-bsl #28: cc2538sf53 write stopping halfway

Johan-Henry, 2015-07-06:
Good day

I am having an issue with your script on my SmartRF06 Eval board with a cc2538sf53 on. Please have a look at the attached pics. Am I doing something wrong?
Erasing works perfectly fine and programming starts off well, but then gets interrupted with the message:
ERROR: No response from target on status request. (Did you disable the bootloader?)

I have tried your script on Linux and OSX on different computers with the same result.
Can you please advise?

Regards
Johan-Henry

[image]

herrfz, 2015-07-10:
Hello, I'm having the same issue on OpenMote. I'm running the script on Linux Mint VM (VirtualBox) on a Windows 7 host.

JelmerT, 2015-07-10:
@Johan-Henry @herrfz It seems that the script doesn't get an answer on the status command after it changed the write address.

Can you still reprogram the chip after this error (is the bootloader disabled)?
Have you tried programming the standard Contiki example, to see if it's a hardware problem or not?
Can you give me the complete command you use to call the script.

herrfz, 2015-07-13:
I think I need to reset, then I'm again able to rerun the script.

I figured in my case it was probably a VirtualBox issue. I tried on my Mac and there was no problem. Here's the command:

    python cc2538-bsl.py -e -w -b 115200 -p /dev/ttyUSB0 --bsl 03oos_openwsn_prog.bin

Johan-Henry, 2015-07-13:
Hi JelmerT

I am using the same command than Herrfz. It seems that my issue might be related to my hardware as I am having a similar issue with another serial programming script written in C++. Sorry for opening this issue, it seems unrelated to your script.

Thanks for the replies

closed 2015-07-13 by Johan-Henry

JelmerT, 2015-07-13:
No problem. I've seen a lot of trouble before with virtual machines and usb-serial drivers / hardware. Adjusting time-outs might help in that case (or changing VM).

RenaKunisaki, 2020-04-05:
I just had this problem with a CC2640. It got as far as 0x0001FF80 before timing out on GetStatus. It seems to have bricked the chip, as it no longer turns on the LED when reset, or responds to UART at all. (I should be able to fix it via JTAG, though.)

The chip is in a custom PCB, so there could very well be a hardware issue, but do you have any ideas?

JelmerT, 2020-04-07:
@RenaKunisaki this sounds like a hardware problem where the serial link timing gets messed up. Are you using a VM like above? Can you replace the usb-uart converter, different pc, ... try to eliminate and swap out until you found which part is making the communication stop.

RenaKunisaki, 2020-04-07:
I'm not using a VM, just Arch Linux. I can try a different converter once I manage to get JTAG working. Do you know what kind of hardware problem might cause this?

JelmerT, 2020-04-07:
Anything between chip and your serial driver on the pc basically.
If you're running a native OS on the pc side, I'd look at the hardware. Maybe try a known good board with the same serial converter, and see if that times-out.

RenaKunisaki, 2020-04-10:
Unfortunately I don't have a known good board. However I've bricked a second chip, under exactly the same condition. It was able to read flash just fine (it spat out all 0xFF, but I assume the flash is empty), then failed writing at 0x1FF800, and now neither UART nor JTAG will respond. I assume it's writing wrong data and flipping the oh-so-handy "security" bits to lock those interfaces...

JelmerT, 2020-04-10:
Can you check your hex file and make sure you arent making any changes to the `CCFG` sector that could possibly have disable any TAPs?

Since your problem seems different from this issue, care to open a new one and describe all the things you've already tried?