JelmerT/cc2538-bsl #31: Locked out after flashing a cc2538 module

marciolm, 2015-08-21:
Hello,
I can't say if is a CC2538 bug, but after flashing a generic CC2538 512k module with the Contiki's udp-client.bin example, I lost the bootloader access (pulling down PA3 and reset). There was previously an other valid image in the flash. The CC2538-bsl doesn't check the bootloader backdoor field before flashing ?  I did something wrong ? I'm afraid in loosing more modules, becaause I dont have the JTAG connection ... 
The procedure in Windows was:
C:\>python cc2538-bsl.py -p com7 -e -w -v client.bin
Opening port com7, baud 500000
Reading data from client.bin
Connecting to target...
    Target id 0xb964, CC2538
Erasing 524288 bytes starting at address 0x200000
    Erase done
Writing 524288 bytes starting at address 0x200000
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x003d1dfb)

C:\>python cc2538-bsl.py -p com7 -e -w -v udp-client.bin
Opening port com7, baud 500000
Reading data from udp-client.bin
Connecting to target...
    Target id 0xb964, CC2538
Erasing 524288 bytes starting at address 0x200000
    Erase done
Writing 524256 bytes starting at address 0x200000
ERROR: No response from target on status request. (Did you disable the bootloade
r?)

JelmerT, 2015-09-02:
The script tries to check if the boot loader backdoor enable bit is set or not and it will tell you if your firmware is about to close the backdoor. But this only works for firmware images of the right size for the 512K chip. (Any help on implementing  check for different sizes and firmware is greatly appreciated)
So, in theory, if your firmware was about to close the backdoor, the script should have caught it.

Make sure, when you compile contiki, that the right bits are set. The standard Contiki config has the bootloader enabled. You can check here: https://github.com/contiki-os/contiki/blob/master/cpu/cc2538/startup-gcc.c

marciolm, 2015-09-03:
I was using an outdated version of Contiki's tree. The current tree at the Github has several switches to control the bootloader field in the ~/contiki/platform/cc2538dk/contiki-conf.h file.
I only spotted this problem with you help. Thank you.
Márcio

closed 2015-09-03 by marciolm