JelmerT/cc2538-bsl #38: how to unlock bootloader if it gets accidentally locked?

lovelesh, 2015-09-15:
Hi @JelmerT ,

I have been using you python script for uploading program to openmotes. But I accidentally changed the target to cc2538dk and now I am not able to unlock the bootloader. I don't get any ACK when I send 0x55 x055 on serial terminal. 

Is there a way out? I am not able to find any web page on this issue. I can get the J LINK SEGGER debugger if required.

Thanks,
Lovelesh

JelmerT, 2015-09-17:
I don't have any Openmotes, so for specific questions about that platform, you'll have to contact them.

That being said: I'm presuming you're using a recent distribution of Contiki, and in the standard CC2538dk platform, the boot loader should be unlocked. You can check this here: https://github.com/contiki-os/contiki/blob/master/platform/cc2538dk/contiki-conf.h

So make sure you're pulling down pin 3 on port A while resetting to get the serial boot loader to start.
If this doesn't work, I'm afraid you'll have to reflash the chip over JTAG with the TI tools.

JelmerT, 2015-10-23:
Did you get this problem solved @lovelesh ?

lovelesh, 2015-10-24:
Hi Jelmer,

I dont have access to the JTAG Debugger. But the TI forum also suggested
the same. So I am buying the Debugger.

Regards,
Lovelesh

JelmerT, 2015-10-27:
Ok, I'll close this issue then. Feel free to comment if you need further help.

closed 2015-10-27 by JelmerT