JelmerT/cc2538-bsl #43: Default baud rate 500k for cc2538 OK?

uknoblic, 2015-10-21:
Hi,

is seems that the default baud rate used for your script is 500000. Did you test this speed with the cc2538? http://www.ti.com/lit/an/swra466a/swra466a.pdf states that the cc2538 only supports 460800.

Regards

Ulf

JelmerT, 2015-10-23:
This speed (500000) does indeed work with the CC2538. Mainly tested by trial and error on the SmartRF06 boards. Never heard any complaints.

Are you seeing problems with a specific platform?

Note that this speed is only used in the beginning of the script, before enabling the external crystal.

uknoblic, 2015-10-23:
Thanks, no problems with this baud rate. As we had the issue with the FTDI converter before using Georges's fix, i was wondering if the baud rate could be the problem. Solved!

closed 2015-10-23 by uknoblic