JelmerT/cc2538-bsl #44: Python problems

uknoblic, 2015-10-21:
Hi,

is there any issue know with the actual python library in ubuntu (14.04)? The bsl-script fails to receive the ACK on the synch sequence in about 50% of the times. We checked that the ACK is send correctly (sending 0x55 0x55 by hand using a windows terminal program). Maybe caused by this issue:

https://bugs.launchpad.net/ubuntu/+source/linux-lts-trusty/+bug/1501345

I hade similar problems with python using another script (https://github.com/g-oikonomou/sensniff)

Regards

Ulf

uknoblic, 2015-10-21:
Sorry,

this topic seems related to issue https://github.com/JelmerT/cc2538-bsl/issues/32. We are using a custom board where the CC2538 is connected to the PC using a FTDI FT234XD.

Regards

uknoblic, 2015-10-21:
https://github.com/JelmerT/cc2538-bsl/pull/40 solves our problem!

closed 2015-10-21 by uknoblic

JelmerT, 2015-10-23:
Perfect! Thanks for confirming #40, I'll merge it into master.