JelmerT/cc2538-bsl #47: CC2650 SBL command fail

juige, 2015-12-24:
Hi all, 

I am using the script to upgrade my CC2650 but got a timeout issue when upgrade firmware. Does anyone have any suggestion?

Following is the log:
[/home/george/Downloads/cc2538-bsl-master]$./cc2538-bsl.py -e -w -v -V -b 115200 -p /dev/ttyUSB0 ./BLEPeripheral.hex 
Opening port /dev/ttyUSB0, baud 115200
Reading data from ./BLEPeripheral.hex
Your firmware looks like an Intel Hex file
For more solid firmware type auto-detection, install python-magic.
Please see the readme for more details.
Connecting to target...
*** sending synch sequence
Got 0 additional bytes before ACK/NACK
*** GetChipId command (0x28)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Version 0x8000F000
    Unrecognized chip ID. Trying CC13xx/CC26xx
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
CC2650 PG2.2 (4x4mm): 128KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x0001FFD8
Primary IEEE Address: 00:12:4B:00:07:EB:75:82
Erasing all main bank flash sectors
*** Bank Erase command(0x2C)
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Erase done
Writing 131072 bytes starting at address 0x00000000
*** Download command (0x21)
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x00000000
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x000000F8

Traceback (most recent call last):
  File "./cc2538-bsl.py", line 1057, in <module>
    if cmd.writeMemory(conf['address'], firmware.bytes):
  File "./cc2538-bsl.py", line 620, in writeMemory
    self.cmdSendData(data[offs:offs+trsf_size]) # send next data packet
  File "./cc2538-bsl.py", line 536, in cmdSendData
    if self._wait_for_ack("Send data (0x24)",10):
  File "./cc2538-bsl.py", line 233, in _wait_for_ack
    % (info,))
CmdException: Timeout waiting for ACK/NACK after 'Send data (0x24)'
ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'

g-oikonomou, 2015-12-24:
Never used with 4x4 but good to see the script detected the package correctly (I hope!?). I cannot see how this could be the cause.

Try without -b please

We may have to increase the timeout to accommodate for this and perhaps even lower speeds. Either that or set a min speed limit (115200 would sound reasonable)

JelmerT, 2015-12-25:
In the past errors like this sometimes were caused by too long wires / unstable serial connections between the chip and the USB-serial converter. 
You could try looking at the signal with a scope, to make sure it's clean. Or slowing down the speed even lower and see if that makes a difference (after you tried without the `-b`).

juige, 2015-12-28:
Thanks all.

The problem is the RESET pin is shared with the SOC. After removing it, the script is work :)

JelmerT, 2015-12-29:
Glad you found the culprit.

closed 2015-12-29 by JelmerT