JelmerT/cc2538-bsl #52: Synch response timeout error with some commands

bthebaudeau, 2016-06-04:
On Zolertia Zoul Firefly, with cc2538-bsl 4340542:
```
$ ./tools/cc2538-bsl/cc2538-bsl.py -V -b 115200 -e
Opening port /dev/ttyUSB0, baud 115200
Connecting to target...
*** sending synch sequence
Traceback (most recent call last):
  File "./tools/cc2538-bsl/cc2538-bsl.py", line 1040, in <module>
    if not cmd.sendSynch():
  File "./tools/cc2538-bsl/cc2538-bsl.py", line 369, in sendSynch
    return self._wait_for_ack("Synch (0x55 0x55)", 2)
  File "./tools/cc2538-bsl/cc2538-bsl.py", line 253, in _wait_for_ack
    % (info,))
CmdException: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```
```
$ ./tools/cc2538-bsl/cc2538-bsl.py -V -b 115200 -a 0x0 -l 0x20000 -r rom.bin
Opening port /dev/ttyUSB0, baud 115200
Connecting to target...
*** sending synch sequence
Traceback (most recent call last):
  File "./tools/cc2538-bsl/cc2538-bsl.py", line 1040, in <module>
    if not cmd.sendSynch():
  File "./tools/cc2538-bsl/cc2538-bsl.py", line 369, in sendSynch
    return self._wait_for_ack("Synch (0x55 0x55)", 2)
  File "./tools/cc2538-bsl/cc2538-bsl.py", line 253, in _wait_for_ack
    % (info,))
CmdException: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

The common condition to these errors seemed to be the lack of input file to read, which made me test if this was a timing issue with the following patch:
```
diff --git a/cc2538-bsl.py b/cc2538-bsl.py
index dd912de..5715d36 100755
--- a/cc2538-bsl.py
+++ b/cc2538-bsl.py
@@ -1035,6 +1035,8 @@ if __name__ == "__main__":
             mdebug(5, "Reading data from %s" % args[0])
             firmware = FirmwareFile(args[0])
 
+        time.sleep(0.1)
+
         mdebug(5, "Connecting to target...")
 
         if not cmd.sendSynch():
```

And it now works. This is not a clean fix, but a only workaround indicating where to look. That's why I have opened an issue and not a pull request for this.

g-oikonomou, 2016-06-04:
See also #50 

I think this is related to the `time.sleep()`s inside `invoke_bootloader`. On the RE-Mote there is a PIC that asserts the SoCs RESET line as soon as it detects the respective sequence on the DTR and RTS lines. I've found out that on some RE-Motes this may take slightly longer than in others. I think what happens is that we are trying to talk to the BSL before it has actually started and if the timing is not deterministic it's very hard to find a suitable value. Perhaps we need to either use a long `time.sleep()` or add a small number of retries if `cmd.sendSynch()` fails.

I'm not 100% certain if the Firefly uses the same design. From what you are reporting, it appears it does. @alignan?

alignan, 2016-06-04:
It uses the same design, I has to increase the sleep time also to 0.1 or alike to be on the safe side, as depending on the OS or VM the flashing failes

arthurfabre, 2016-06-13:
Same issue here using a remote on a baremetal ubuntu 16.04 install (but works fine on a VM).
Changing both the sleeps from `0.002` to `0.02` resolves the issue on the physical machine.
Is everyone happy for me to submit a pull request making this change?

g-oikonomou, 2016-06-13:
Yes, a PR is welcome. But I think we need to be certain about the correct wait time here; this is something we thought we had fixed already.

alignan, 2016-06-13:
`0.02` was a bit too optimistic to begin with... `0.1` should be pretty safe

bthebaudeau, 2016-06-13:
`0.1` for both sleeps works fine for me with Zoul Firefly on Lubuntu 16.04.

arthurfabre, 2016-06-17:
As far as I can tell from the comments, we shouldn't need to increase the 1st timeout - it seems to be a Mac specific thing (commenting it out on Ubuntu 16.04 doesn't break it either).

`0.1` works for me too with Remotes on Ubuntu 16.04.

JelmerT, 2016-06-18:
closed by #53

closed 2016-06-18 by JelmerT