JelmerT/cc2538-bsl #55: Support for MSP432?

peretuset, 2016-07-01:
@JelmerT I have read through the MSP432 (http://www.ti.com/lit/ug/slau622a/slau622a.pdf) and compared to the CC2538/CC2650 (http://www.ti.com.cn/cn/lit/an/swra466a/swra466a.pdf) documentation, and the BSL implementation seems different. However, the communication mechanisms between the chip and the PC are similar, so I am wondering if you/someone would be interested in adding support for the MSP432 chip to the current implementation.

JelmerT, 2016-07-06:
When we added support for the CC2650 we specifically added some abstraction with the idea to support more chips in the future. So I'm all for, this and I welcome all pull requests.

I don't have any MSP432 HW though, so I currently can't work on this myself.

g-oikonomou, 2016-07-06:
Need a little more work on that abstraction, but this could be an excellent opportunity to do this. Depends how close the msp instruction set is to what we have already. Is the actual packet format the same?

JelmerT, 2016-07-07:
Looks like there are some significant differences:
(left msp, right CC)
[image]

JelmerT, 2017-10-05:
Pull requests on this still very welcome, but closing this issue for now.

closed 2017-10-05 by JelmerT