JelmerT/cc2538-bsl #56: Restart the device via command line

alignan, 2016-07-06:
Hello!

Something that would be really cool and handy would be to restart a device over USB, is this feature available, or on the roadmap?

Thanks!

JelmerT, 2016-07-06:
When you're referring to restarting the chip, I'm presuming you mean restarting while it's running the flashed image. This might be difficult since you can only send a reset command when you're actually communicating with the bootloader (which has to be started manually).

If the hardware supports the RTS (serial pin) to RST (pin on the chip) mapping, then it wouldn't be too hard to add this I think. 
How does the PIC on the Re-Mote handle the incoming RTS signal?

niolp, 2016-09-19:
Hi Jelmer.
Do you mean RTS pin on an FTDI232R device? Are there any other hardware considerations to make this work?
Regards

JelmerT, 2016-09-27:
@niolp yes the RTS on an FTDI or similar USB to serial chip. The idea is to connect the RTS or DTR of the serial line to the reset pin on the chip and be able to hard reset the chip that way. You can check the `invoke_bootloader` function for more details.

Tanganelli, 2018-01-11:
Hi Jelmer,
I am using a zoul board and the contiki OS. I tried to define a reset function inspired by your `invoke_bootloader`. The following is the result:

`class CommandInterface(object):
    ACK_BYTE = 0xCC
    NACK_BYTE = 0x33
    def open(self, aport='/dev/tty.usbserial-000013FAB', abaudrate=500000):
        self.sp = serial.Serial(
            port=aport,
            baudrate=abaudrate,     # baudrate
            bytesize=8,             # number of databits
            parity=serial.PARITY_NONE,
            stopbits=1,
            xonxoff=0,              # enable software flow control
            rtscts=0,               # disable RTS/CTS flow control
            timeout=0.5             # set a timeout value, None for waiting forever
        )

    def reset(self, dtr_active_high=False, inverted=False):
        if inverted:
            set_reset_pin = self.sp.setDTR
        else:
            set_reset_pin = self.sp.setRTS
        set_reset_pin(0)
        set_reset_pin(1)
        set_reset_pin(0)
        time.sleep(0.2)
        `
However, it doesn't work. Do you have any suggestion?

Regards
Giacomo

JelmerT, 2021-01-20:
I guess this feature was implemented a while ago. Needs HW support of course.
Closing!

closed 2021-01-20 by JelmerT