JelmerT/cc2538-bsl #57: ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

AmitSharma1094, 2016-08-02:
Hi,

I am using srf06EB. I am trying to flash cc1310 with contiki demo code. I gettign ACK-NACK error.

PS:- I already modified BL setting as per suggestion mention in the README.

Command:-
`sudo make TARGET=srf06-cc26xx BOARD=srf06/cc13xx cc26xx-demo.upload`

Error:-
```
python ../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -p /dev/ttyUSB0 cc26xx-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from cc26xx-demo.bin
Firmware file: Raw Binary
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
make: *** [cc26xx-demo.upload] Error 1
```
 

Br
Amit Sharma

closed 2016-08-02 by AmitSharma1094

JelmerT, 2016-08-02:
@AmitSharma1094 you closed this, did you fix your issue? If yes, could you share your error or solution for future reference?

AmitSharma1094, 2016-08-03:
Hi Jelmer,
 
Yes it's fixed. 

Solution :- i was flashing the code on wrong Port[I was flashing the code on ttyUSB0, but later i changed it to ttyUSB1].

For reference please refer below mention link.
[](http://e2e.ti.com/support/wireless_connectivity/proprietary_sub_1_ghz_simpliciti/f/156/p/529953/1941134#1941134)
 
Br
Amit Sharma

AmitSharma1094, 2016-08-03:
Sorry, forget to add link

http://e2e.ti.com/support/wireless_connectivity/proprietary_sub_1_ghz_simpliciti/f/156/p/529953/1941134#1941134

JelmerT, 2016-08-03:
@AmitSharma1094 👍

nusrathmouri, 2018-11-14:
Hi Jelmer,
I am in the similar trouble. When I want to upload cetic_6lbr_router_eth_gw.bin image in Orion router it shows the above mentioned error: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'. But the same process is exactly working for zolertia re-mote device but it is not needed. Even if I am changing ports these are not fixing the problem in orion router. Could please give some suggestions?

JelmerT, 2018-11-15:
@nusrathmouri this means the script isnt getting any response from the micro in the orion router. Make sure all your connections work, also have a look with @zolertia on how to reflash the router correctly.

nusrathmouri, 2018-11-16:
Hi,
Now the previous problem is solved. Can you please tell me what is the
errors I need to solve for uploading mqtt-client file(attached output
screenshot). I need to solve it by tomorrow. Please help me.

Nusrath

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
16/11/18,
13:57:51

JelmerT, 2018-11-16:
So what was the solution to your previous problem? Please share so everyone can learn.
If you're having a different issue, please open a new one.

nusrathmouri, 2018-11-19:
May be the problem was with connectivity. When I close my VMware and started again it was working correctly. But after uploading cetic_6lbr_router_eth_gw binary image, I am facing new problem with router configuration and mqtt-client also has some errors.

LuighiV, 2019-12-11:
> Sorry, forget to add link
> 
> http://e2e.ti.com/support/wireless_connectivity/proprietary_sub_1_ghz_simpliciti/f/156/p/529953/1941134#1941134

That article is pretty interesting beacuse it explains that to use the script it is required that the chip has the bootloader enabled. Something important to highlight is that the sample image provided with chips doesn't have this mode enabled by default. So, for the first time, it is required to flash them using the Smart RF Tool or the Uniflash tool http://www.ti.com/tool/download/UNIFLASH  with a setting with booloader enabled. For example, with the following code:

 #define ROM_BOOTLOADER_ENABLE 1

In the project settings. And after it, you can load any other program using the bootloader script provided in the contiki repository.

Hope it could be useful to whom in trouble with this setting.
Best regards,