JelmerT/cc2538-bsl #59: Add licence

gfarrell, 2016-10-13:
Great script, could you add a licence to it so it's clear? Looking at integrating into some of our tooling.

kgraefe, 2018-06-27:
According to the [script header](https://github.com/JelmerT/cc2538-bsl/blob/master/cc2538-bsl.py#L3) the script is licensed under [The 3-Clause BSD License](https://opensource.org/licenses/BSD-3-Clause).

Hedda, 2022-01-10:
@JelmerT @flatsiedatsie posted a question in https://github.com/JelmerT/cc2538-bsl/pull/114 asking if cc2538-bsl can really be ``AS IS'' licenced if based on stm32loader?

I guess real question is if cc2538-bsl directly copies any lines from stm32loader? or only takes inspiration or revense engineerd?

https://github.com/JelmerT/cc2538-bsl/blob/master/cc2538-bsl.py

```
# Implementation based on stm32loader by Ivan A-R <ivan@tuxotronic.org>
```

"stm32loader" code in these repositories looks to be licensed under GPLv3 (GNU General Public License version 3):

https://github.com/jsnyder/stm32loader

https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/flash/stm32loader.py

If it actually copies full lines of codes then there is a point where it would count as needing to use a compatible license or?

Hedda, 2022-01-11:
> @JelmerT @flatsiedatsie posted a question in #114 asking if cc2538-bsl can really be ``AS IS'' licenced if based on stm32loader?
> 
> I guess real question is if cc2538-bsl directly copies any lines from stm32loader? or only takes inspiration or revense engineerd?
> 
> https://github.com/JelmerT/cc2538-bsl/blob/master/cc2538-bsl.py
> 
> ```
> # Implementation based on stm32loader by Ivan A-R <ivan@tuxotronic.org>
> ```
> 
> "stm32loader" code in these repositories looks to be licensed under GPLv3 (GNU General Public License version 3):
> 
> https://github.com/jsnyder/stm32loader/blob/master/stm32loader.py
> 
> https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/flash/stm32loader.py
> 
> If it actually copies full lines of codes then there is a point where it would count as needing to use a compatible license or?
> 
> Best would regardless be to add a separate e `LICENSE` or `LICENSE.md` (with all caps) file to this repository. See:
> 
> https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-a-license-to-a-repository

Hopefully no code is copied and can merge https://github.com/JelmerT/cc2538-bsl/pull/117

JelmerT, 2022-01-20:
"loosely based" means it was looked at for the structure. It was the only other serial bootloader script in Python I could find at that time. So we can keep BSD-3.
I'll pull #117 which closes this issue.

closed 2022-01-20 by JelmerT