JelmerT/cc2538-bsl #61: Can not Flashing a 6lbr image to the Zolertia Ethernet IoT Gateway

zhangtuoparis13, 2016-12-08:
Hi all,

  I follow the instruction on [Zolertia/Resources/wiki/6lbr](https://github.com/Zolertia/Resources/wiki/6lbr)
  then I have a problem while flash the 6lbr image to the  Zolertia Ethernet IoT Gateway.
```
user@ubuntu:~/6lbr/examples/6lbr$ ls -l /dev/ttyUSB0 
crw-rw---- 1 root dialout 188, 0 Dec  8 03:50 /dev/ttyUSB0
user@ubuntu:~/6lbr/examples/6lbr$ python ../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v bin_zoul/cetic_6lbr_smart_bridge.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from bin_zoul/cetic_6lbr_smart_bridge.bin
Connecting to target...
ERROR: Can't connect to target. Ensure boot loader is started. (no answer on synch sequence)
```
i have also tryed with config the -p /dev/ttyUSB0  and -b 115200, it does not work either.

Is there anyone expertise could help me to solve the problem?

Thanks a lot!

ps: but it can be flashed with the contiki/examples/hello-world as a zoul when ` make TARGET=zoul MOTES=/dev/ttyUSB0 hello-world.upload login `

g-oikonomou, 2016-12-08:
@alignan any comment here?

alignan, 2016-12-08:
This is the wrong place to ask for help, as it depends on the 6lbr version of the script, which at the date I think is some commits behind.  You could always do a fresh clone of the cc2538-bsl repository, and just flash the compiled binary, even try to replace the cc2538-bsl.py file.

I will return on December 29th, until then good luck!

zhangtuoparis13, 2016-12-08:
all my clones are fresh, it doe not work even replace the cc2538-bsl.py file.
i will continue to figure the issue.
@alignan good holiday, thx anyway~

zhangtuoparis13, 2016-12-08:
yes it works when i do a fresh clone of the cc2538-bsl repository. thank you @alignan .

closed 2016-12-08 by zhangtuoparis13