JelmerT/cc2538-bsl #64: Write corruption

splitice, 2017-08-14:
How to replicate: 
```
halley@Tinyhub:~$ python ~/cc2538-bsl/cc2538-bsl.py -p /dev/ttyS1 -b 115200 -w ZNP.bin
Opening port /dev/ttyS1, baud 115200
Reading data from ZNP.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 00:12:4B:00:10:05:FF:DB
Writing 503808 bytes starting at address 0x00200000
Write 120 bytes at 0x0027AF888
    Write done
halley@Tinyhub:~$ python ~/cc2538-bsl/cc2538-bsl.py -p /dev/ttyS1 -b 115200 -r read.bin
Opening port /dev/ttyS1, baud 115200
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 00:12:4B:00:10:05:FF:DB
Reading 524288 bytes starting at address 0x200000
    Read done

```

Comparison example:
[image]

We tested different baud rates both slower and faster including the default with no changes.

JelmerT, 2019-01-04:
You're missing the length of the data you want to read (`-l` option). The `-r` option by default reads `0x80000` bytes of data. But it looks like your firmware has a different length.

The read function could use some general improvement, it was never thoroughly tested. But we could add a warning in the script for now that notes that it is using the standard read length.

splitice, 2019-01-04:
I think you missed the point. Anyway we were able to resolve this with our fork, we rewrote substantial portions of the communication code (although our fork is CC2538 only).

closed 2019-01-04 by splitice

JelmerT, 2019-01-05:
Care to elaborate what the point exactly was, or point to your fork so we can improve this script?