JelmerT/cc2538-bsl #73: cc253x support

itsmepetrov, 2018-08-23:
Any chance of adding support for cc253x?

g-oikonomou, 2018-08-23:
Have you checked cc-tool?

https://github.com/scott-42/cc-tool
https://github.com/dashesy/cc-tool

Not sure which one is newer

itsmepetrov, 2018-08-23:
Yes, I've already checked, but I haven't seen any support for `.bin` images for `sbl` loader. As I understand it works only for full images (.hex) and only via `cc-debugger`.

g-oikonomou, 2018-08-23:
I personally don't plan to work on adding support for that chip family, no.

cc-tool uses the debug interface last time I checked, yes. As far as I can tell, this sbl is not on chip ROM, is it? So one has to build it and then program the device with it, it sits in the low area of flash. If one can build it then one can certainly create a combined hex/bin that contains the bootloader + the main application.

If the protocol is similar to the one used by CC2538 then it should not be too big effort to support if you were to decide to try.

JelmerT, 2019-01-04:
Closing this for now, support for CC2530/31 would be great, but it's not an easy add-on, and there's no one volunteering at this moment. 
This ties in to the previous idea of expanding the script into a universal tool for all chips with a serial bootloader flashing mechanism.

Pull requests are welcome though!

closed 2019-01-04 by JelmerT