JelmerT/cc2538-bsl #78: CC1350 sensor tag

toothless90x, 2019-01-02:
Is there a way to hardwire CC1350 sensor tag to interact with this script ?

g-oikonomou, 2019-01-03:
Absolutely not for the CC2650 sensortag, since the relevant pins are not exposed. I suspect (but have not checked) that the same applies for the CC1350 one

JelmerT, 2019-01-04:
TI might actually improved the design on this version a (tiny) bit! 
I don't have the hardware but if I look at the design files it looks like the bootloader UART pins are on DIO2 and DIO3 (7x7QFN). Looking at the schematic of the sensortag these pins are used for the digital mic, but they conveniently added 2 jumpers on those traces (R21 and R22).
[image]
[image]

So if you're willing to give up the mic, and have a lot of soldering fun, you might actually be able to use the UART bootloader on the CC135 sensortag. I'm pretty sure that when the mic sees a clock on the `audio_clk` pin it's going to dump data on the `audio_di` pin, so just hot-wiring it probably won't work (but worth a try, might wreck the mic though).

Please try and report back!

akrv, 2019-01-04:
Yes, this works. I have tested it.
But, must be removed for Microphone to work.
we are using a [TMUX12095-VBidirectional 4:1, 2-Channel Multiplexer](http://www.ti.com/product/TMUX1209) to mux the pins with the APP_UART and BL_UART.

JelmerT, 2019-01-04:
Great!

Thanks for confirming @akrv !

JelmerT, 2019-01-04:
Something in the readme about this might be good.

closed 2019-01-04 by JelmerT