JelmerT/cc2538-bsl #83: CC1350 sensortag : write error "Send data (0x24)"

toothless90x, 2019-02-11:
The device is hardwired for DIO 2 & 3 for Bootloader UART comm. 
While trying to flash the board using the script, the following error turns. 
terminal log also attached for the reference. 
How to solve this ? 

[screen1.log](https://github.com/JelmerT/cc2538-bsl/files/2851260/screen1.log)

JelmerT, 2019-02-11:
Can't see the log you linked

toothless90x, 2019-02-11:
Its the same file. I made sure it opens in any notepad. 
[screen1.log](https://github.com/JelmerT/cc2538-bsl/files/2851815/screen1.log)

JelmerT, 2019-02-11:
My bad, addblocker was a little too blocky

JelmerT, 2019-02-11:
Does the error always come at the same time? It looks like communication is working, but then randomly fails.
Did you remove R21 and R22 on the board? And you properly connected ground?

toothless90x, 2019-02-11:
yes. always at happens.
Those R21 and R22 are removed for DIO2,3 connection. 
Ground is connected properly.

JelmerT, 2019-02-11:
Just want to make sure the digital mic isn't interfering with the UART.

Not sure about this one. Have you tried a different source file? Maybe smaller to see what difference that that makes. And you could try slowing down the baudrate.

toothless90x, 2019-02-12:
For different source file, when I try to make .bin for hello-world with following cmd in git clone
`make TARGET=srf06-cc26xx BOARD=sensortag/cc1350 hello-world.bin`
the .bin file size is 131.1kb. The interesting factor is, for any test example when .bin is made, the file size is again 131.1 kb. 
I dont know why it happens like this .

JelmerT, 2019-02-12:
I don't have contiki-ng setup right now, so I can't check. But can you try a different project that isn't hello-world?

g-oikonomou, 2019-02-12:
I would be very surprised to see a .bin file with a size other than exactly 128KB :)

g-oikonomou, 2019-02-12:
To the topic in hand, are you sure you are not simply falling victim to dodgy wires? What are you physical connections like? What do you have connected to what?

toothless90x, 2019-02-12:
Got it resolved. It was an voltage fluctuation from MOSFET (Used to power reset the Board-CC1350), which interrupts the Send data inbetween the process. 

Connection: Host PC - FTDI - hardwired sensortag CC1350(mounted on custom made Mother board).

@JelmerT  Its turns out to be same size .bin (131.1Kb) for cc26xx-web-demo example also. :(
@g-oikonomou Regarding the .bin file issue, shall I open a new thread ?

g-oikonomou, 2019-02-12:
The bin file size is not an issue, it's perfectly expected. Just don't try to estimate firmware code footprint by looking at the file size of the .bin. Use `arm-none-eabi-size` on the .elf. No need to open a separate thread.

JelmerT, 2019-02-12:
Great, looks like the problem was found, closing this.

Thanks for the help @g-oikonomou

closed 2019-02-12 by JelmerT