JelmerT/cc2538-bsl #84: ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)' With OPENMOTE-B and RIOT

mohanmaniramesh, 2019-02-20:
Hello all,
I'm getting this   Error
**ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'**
when am flashing the code second time
I will explain step by step:

1. I have 10  openMote-B  Rev-1.4
2. I started working with RIOT. I'm confused where to and how to start so I mailed to RIOT Developer's
   they said  as you are a beginner you can  try  RIOT--TUTORIALS like that
   **https://github.com/RIOT-OS/Tutorials**
3. So I started working with the TUTORIALS, I have installed the necessary packages
4. I have connected only one OpenMote-B to my PC having ubuntu18.04
5. I went to this Directory     /home/Tutorial/Task-01
6. I Provided This Comment  ** BOARD=openmote-B make flash all term**  For tranfer the program into
   my openmote-B
7. The Code was transferred successfully and a terminal Opened there I gave PS, Reboot  and Help
   Command that was also working well
8. Now i went the Task-02 Directory     /home/Tutorial/Task-02
   Here I gave this command  **BOARD=openmote-B make flash all term** For tranfer the TASK-02 program to my openmote-B
   But this Time I got the Following Error                                                                                                                                                     
                                                  **ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'**

I got to know by my analysis there is problem with the Bootloder kind of things only
openMote-B developer Mr.per tuset he said the following
Hello,
You have probably flahsed a RIOT image that has the wrong pin configuration for the bootloader in the Flash. To fix it:

1. Use a JTAG probe to flash an empty image or am image with the proper bootloader Flash settings
2. Change the RIOT image to make sure that the same bootloader Flasg setting are used
   For your reference, the pin that is used to trigger the bootloader on the OpenMote-B board is **PORTA PIN7** using a low level.
   Kind regards,
   Pere
   I analysed and found the below line
   #define CCA_BACKDOOR_PORT_A_PIN   (6) /**< BSL_BOOT Pin */
   by Pere Tuset approval i changed the pin number 6 into 7

but then also not working and again he said like this
Hello,
This is really unfortunate.
I am not very familiar with RIOT, but there must be something else going on.
For example the CCA configuration is stored in Flash as described in this file -> **https://github.com/RIOT-OS/RIOT/blob/master/cpu/cc2538/vectors.c**
The specific line is : **0xe0ffffff | (CCA_BACKDOOR_ENABLE << 28) | (CCA_BACKDOOR_ACTIVE_LEVEL << 27) | (CCA_BACKDOOR_PORT_A_PIN << 24)**
If done the math and it matches the configuration I am unsing on my projects, so that should work.
One thing to try is using a slower baudrate (i.e. 400000) just in case.

but I'am confused where to change what to change i did so many thing
i have total 10 in that 5 boards are locked remaining 5 boards are there i'm fearing to check more things
please help me out of this

(I got the idea to recover the other 5 openmote-B board using JTAG programmer I'm going to make it work)

But i sould work with other boards by solving this error 
please help to solve this error

Thanks

JelmerT, 2019-02-28:
@mohanmaniramesh did you fix your problem with a JTAG programmer?
With the JTAG programmer you can find the right config until the bootloader works

mohanmaniramesh, 2019-02-28:
yes i did with the JTAG. but i didn't  solved  my bootloader  i'm
completely  using JTAG only for flash the program. what are modifications
i have to do , to use BSL programming  please help me

JelmerT, 2019-02-28:
this is where the correct bits are set for the openmoteB https://github.com/RIOT-OS/RIOT/blob/master/boards/openmote-b/include/board.h

I can't find the schematics of the openmoteB, so I can't check them, but I'm assuming these are correct. (pin counting starts from 0) So try to flash a mote with the standard config and test the bootloader.

mohanmaniramesh, 2019-02-28:
Ok I'll  replace the board.h file
and Am attaching openMote-b Schematics
for your reference
Thank you

JelmerT, 2019-12-22:
Closing this, feel free to re-open if still having issues

closed 2019-12-22 by JelmerT