JelmerT/cc2538-bsl #85: CC1352 support

Gateworks, 2019-03-12:
I've tried this script on a CC1352P and found the following:
- device_id:0x2bb4102f
- wafer_id:0xbb41
- pg_rev:2
user_id:0x2282f00
rev_minor=0xff
FLASH_SIZE=0x2c

In the case of the CC1352 the FLASH_SIZE sector size is 8192 instead of 4096 so the following patch would make this work for CC1352 but I'm sure that would break some other CC13xx variant. What is the right way to detect the difference between CC1352 and whatever other CC13xx variant that has a sector size of 4096?

@@ -828,15 +832,23 @@ class CC26xx(Chip):
         # We can now detect the exact device
         if wafer_id == 0xB99A:
             chip = self._identify_cc26xx(pg_rev, protocols)
-        elif wafer_id == 0xB9BE:
+        elif wafer_id == 0xB9BE or wafer_id == 0xBB41:
             chip = self._identify_cc13xx(pg_rev, protocols)
 
         # Read flash size, calculate and store bootloader disable address
         self.size = self.command_interface.cmdMemReadCC26xx(
-                                                 FLASH_SIZE)[0] * 4096
+                                                 FLASH_SIZE)[0] * 8192

JelmerT, 2019-03-17:
Thanks for figuring this out and reporting back!

If all CC1352 have this wafer_id and the same sector size, we can use the `wafer_id` to select the sector size.
Feel free to submit a pull for this, if I find some time I can add it and I'll ping back here for you to test.

omerk, 2019-12-10:
Added a few notes about this on #88

JelmerT, 2019-12-22:
Closing this after #88 thanks, @Gateworks !

Opening a new issue related to updating the function names to also encompass CC26x2

closed 2019-12-22 by JelmerT