JelmerT/cc2538-bsl #86: ERROR: string argument without an encoding

sebbad, 2019-03-20:
Running cc2538-bsl on Mac OS with python 3.7 provides the following errors:

"ERROR: string argument without an encoding"

This seems to be related to the handling of bytearrays in recent python versions. If I understand it correctly recent python version require the encoding to be explicitly stated in the bytearray function if the parameter is a string. This seems to be the case in the script. When providing an encoding (tried with 'utf8'), the error disappeared.

JelmerT, 2019-03-26:
Thanks for reporting this, interesting that this hasn't come up earlier.

Feel free to submit a PR adding the encoding.

JelmerT, 2021-01-20:
@sebbad could you describe the steps to reproduce this bug?

felipefbs, 2021-04-21:
I have the same issue here. I run through all the c2538-bsl.py code and identified in which line (139) the problem is.

I'm gonna open a PR soon :)

felipefbs, 2021-04-21:
I had this same issue, and I run through the cc2538-bsl.py code and found that the error it's on the byte array casting. But I found that I was running an old version and using the new version I had no problems. Seems that everything its good now.
```python

if have_magic:
            file_type = bytearray(magic.from_file(path, True))
## I just put the "utf-8" and worked for me
if have_magic:
            file_type = bytearray(magic.from_file(path, True), "utf-8")
```

JelmerT, 2021-04-30:
@sebbad can you confirm your issue is fixed with the latest version of the script?