JelmerT/cc2538-bsl #89: Publish on PyPI?

omerk, 2019-12-10:
With a whole bunch of new TI parts supporting BSL (also a good time to rename?), it would be really handy if this script could be installed with a simple: `pip install cc2538-bsl`

A very similar example is the [stm32loader](https://pypi.org/project/stm32loader/) project.

Not a Python expert but I can have a go at modularising and adding the necessary bits if this is something you would consider?

JelmerT, 2019-12-22:
Pretty sure it's already on there: https://pypi.org/project/cc2538/
Not maintained by me though, think it's maintained by @remyleone .

And yes, we should rename. Any suggestions? `Ti-bsl`, `CC-bsl`?

mfaccin, 2020-11-25:
IMHO `cc-bsl` or even `cc-loader` are good choices. :)

g-oikonomou, 2020-11-25:
cc-bsl sounds good. In the past I have also thought of simple-bsl

omerk, 2020-11-25:
Another vote for cc-bsl here

JelmerT, 2020-11-26:
Lets put it to a vote! please ll vote in #103 
Outcome doesn't mean it will become that name, this is just to see if we have a clear winner.

omerk, 2021-01-20:
With the new name established in #103, it would be great if the PyPi package is updated/renamed once repo is renamed.

If you need any volunteers for upkeep etc I am happy to spend some time on this @JelmerT

Hedda, 2022-01-20:
@JelmerT Any updates/news on this?

Can we now please request that you push a new release for PyPI using the old or new name?

https://github.com/JelmerT/cc2538-bsl/issues/89

https://github.com/JelmerT/cc2538-bsl/releases

Also suggest that you use releases with release version in addition to tags:

https://github.blog/2013-07-02-release-your-software/

Plus make use of the PyPI publish GitHub Action

https://github.com/marketplace/actions/pypi-publish

Not sure if project cc2538 there is still the primary as only @Sieben is listed as maintainer there? 

https://pypi.org/project/cc2538/

Hedda, 2022-04-19:
@JelmerT Any news on updating releases on PyPI?

JelmerT, 2022-04-19:
I'm not maintaining this one https://pypi.org/project/cc2538/ that would be @sieben. If there's no answer from them I'm open to maintaining a new package under the new name.

I'll implement the name change somewhere next 2 weeks when I find some time in between my day-job.

remyleone, 2022-04-19:
Hello, I'm @remyleone, @sieben is my old GitHub profile. I'm willing to give ownership of this package to whoever wants it. What is your PyPI username?

JelmerT, 2022-04-20:
Thanks @remyleone ! you can add https://pypi.org/user/jelmt/

remyleone, 2022-04-20:
Done. Feel free to remove me once you accepted the invitation :)