JelmerT/cc2538-bsl #95: Failed flashing/bricking CC2640

RenaKunisaki, 2020-04-11:
I've designed a [custom board](https://easyeda.com/editor#id=2955f4a7a5fc4deeb59646e894aad25a)* using the CC2640, and tried twice to use this script to program it. In both cases, it was able to read flash OK, but failed during the flashing process:

```
sudo ./cc2538-bsl.py -w /home/rena/projects/ccstudio/hello_CC2650DK_7ID_TI/Debug/hello_CC2650DK_7ID_TI.hex -p /dev/ttyUSB0
[sudo] password for rena: 
./cc2538-bsl.py:971: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if int(value) % int(device.page_size) is not 0:
./cc2538-bsl.py:976: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if int(value, 16) % int(device.page_size) is not 0:
Opening port /dev/ttyUSB0, baud 500000
Reading data from /home/rena/projects/ccstudio/hello_CC2650DK_7ID_TI/Debug/hello_CC2650DK_7ID_TI.hex
Firmware file: Intel Hex
Connecting to target...
CC2640 PG2.3 (5x5mm): 128KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x0001FFD8
Primary IEEE Address: FF:FF:FF:FF:FF:FF:FF:FF
Writing 131072 bytes starting at address 0x00000000
Write 128 bytes at 0x0001FF800
ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'
```

This seems to have also locked or damaged the chips, as they now neither turn on the LED (DIO_9) at startup nor respond to UART or JTAG. (Tried with two boards, same result both times. TDO and TX just remain high. JTAG was working before flashing.)

The first time, I mistakenly flashed an ELF file; the second time, I flashed the attached Hex file, which is generated by CCStudio. I'm still not really sure if that's the correct file to be flashing; it could be entirely my mistake, although the fact that it times out, and doesn't warn that I'm about to lock the bootloader, makes me suspect there is a bug as well.

I have three more boards and need at least one of them working, so I'm reluctant to try flashing any more, but I'll keep trying to find a way to get JTAG working again.

Of note, although reading flash didn't generate any errors, it was very slow and produced a file containing all 0xFF. I assume this is just because the flash was empty.

OS: Arch Linux AMD64
Serial adapter: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC at 3.3V
JTAG adapter: https://github.com/myelin/teensy-openocd-remote-bitbang

* unfortunately I forgot to save a backup of the actual board before making changes, so this version is slightly different and has some missing GND connections. I might be able to recover the original design.

RenaKunisaki, 2020-04-11:
http://ix.io/2hCV apparently I can't attach the hex file  here

JelmerT, 2020-04-16:
The link to your board doesnt seem to be working.
Since you seem to have a working JTAG interface, can you try flashing a known good hex file on a new board using jtag? This would clear up a problem with the binary or the script.

You mention reading the flash produces an empty file, but also mention that a LED used to turn on, so was there frimware on the chip before or not? Since without firmware, the led shouldnt turn on.

JelmerT, 2020-04-16:
Looks to me like your firmware disables the PRCM TAP, which probably would have an influence on jtag and bootloader. Not sure about this register location though, haven't worked with CC26xx in a while.
[image]

RenaKunisaki, 2020-04-17:
Interesting. It looks like it put 0xC500C5FF at 0x1FFE4, but `BOOTLOADER_ENABLE` is at 0x1FFD8 and should be 0xC500xxC5. I'm totally new to TI's stuff so I don't know why it would have done that. I wonder if this hex file has some kind of header that's not meant to be sent to the chip?

I think the LED being turned on is a side effect of the bootloader (which isn't in flash), as it configures DIO9 (my LED pin) as MOSI.

I don't have a known good hex file. The JTAG adapter I'm using, it seems to work with OpenOCD (I can read out RAM) but trying to flash with it fails. (Might just be I'm doing something wrong. I've never used JTAG before.) It's also not supported by ccstudio/SmartRF. I'm working on getting one that is.

I assume the reason the flash fails is that it's sending more data than the chip expects? So as far as your code is concerned, the only bugs would be that and the failure to warn about locking the bootloader. I appreciate the help, though!

JelmerT, 2020-06-11:
Checking the bootloader config bits on the input file would indeed be a very welcome feature. But It can get pretty tricky and potentially annoying too. Pull requests are always welcome!

Closing this for now, let me know if you're still having issues.

closed 2020-06-11 by JelmerT

RenaKunisaki, 2020-06-11:
I only mentioned checking the bootloader bits because I thought that was already an implemented feature that had somehow failed. Sorry for any confusion.