JelmerT/cc2538-bsl #96: how to use --bootloader-invert-lines ?

fengjikui, 2020-06-03:
hello JelmerT:
I am using you code to flash the image to cc2650 LaunchPad, I run the cmd like this:
'''python ../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -b 115200 -e -w -v build/cc26x0-cc13x0/launchpad/cc2650/hello-world.bin --bootloader-invert-lines'''

it work not so good:
Opening port /dev/tty.usbmodemL10002881, baud 115200
Reading data from build/cc26x0-cc13x0/launchpad/cc2650/hello-world.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

and if I do what you say to start boot loader by *select + reset*, it works well.
so how can I invoke the bootloader without togglling any pins?
I am very thank you for your any reply.
@JelmerT

JelmerT, 2020-06-11:
To auto start the bootloader via the serial port, you need to have hardware support. the DTR and RTS pins of your serial port need to be connected to the reset and bootloader pins on your chip.
I'm not sure if this is the case on the Launchpad, but I'm guessing this is your issue.

closed 2020-06-11 by JelmerT

fengjikui, 2020-06-11:
thank you so much, I will check it!