JelmerT/cc2538-bsl #100: Fails to execute on MacOS Catalina cc2538-bsl.py:971: SyntaxWarning: "is not" with a literal. Did you mean "!="?   if int(value) % int(device.page_size) is not 0:

wojciechczyz, 2020-10-07:
python cc2538-bsl.py -h
/Users/w/cc2538-bsl/cc2538-bsl.py:971: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if int(value) % int(device.page_size) is not 0:
/Users/w/cc2538-bsl/cc2538-bsl.py:976: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if int(value, 16) % int(device.page_size) is not 0:
Traceback (most recent call last):
  File "/Users/w/cc2538-bsl/cc2538-bsl.py", line 59, in <module>
    from intelhex import IntelHex
  File "/Users/w/.pyenv/versions/3.9.0b5/lib/python3.9/site-packages/intelhex/__init__.py", line 44, in <module>
    from intelhex.compat import (
  File "/Users/w/.pyenv/versions/3.9.0b5/lib/python3.9/site-packages/intelhex/compat.py", line 60, in <module>
    array_tobytes = getattr(array.array, "tobytes", array.array.tostring)
AttributeError: type object 'array.array' has no attribute 'tostring'

JelmerT, 2020-10-07:
This looks like an issue in the intelHex library, I can't reproduce it though.

Could you check your intelHex lib version with

```
$ pip3 show intelhex 

Name: intelhex
Version: 2.2.1
Summary: Python library for Intel HEX files manipulations
Home-page: https://pypi.python.org/pypi/IntelHex
Author: Alexander Belchenko
Author-email: alexander.belchenko@gmail.com
License: BSD
Location: /usr/local/lib/python3.8/site-packages
Requires:
Required-by: pyocd
```

JelmerT, 2020-10-07:
the 2 SyntaxWarnings should be fixed with 733e6f5

wojciechczyz, 2020-10-07:
Hi,

I have the same version you have got.
Python is 3.9.05b.
Strange that intelHex library is triggereed when I am just doing -h , can we avoid it?

Cheers,
Wojciech

wojciechczyz, 2020-10-08:
Thank you for the correction, now I am getting only this:
python cc2538-bsl.py -h

Traceback (most recent call last):
  File "/Users/w/cc2538-bsl/cc2538-bsl.py", line 59, in <module>
    from intelhex import IntelHex
  File "/Users/w/.pyenv/versions/3.9.0b5/lib/python3.9/site-packages/intelhex/__init__.py", line 44, in <module>
    from intelhex.compat import (
  File "/Users/w/.pyenv/versions/3.9.0b5/lib/python3.9/site-packages/intelhex/compat.py", line 60, in <module>
    array_tobytes = getattr(array.array, "tobytes", array.array.tostring)
AttributeError: type object 'array.array' has no attribute 'tostring'

wojciechczyz, 2020-10-08:
Remaining problem is already reported in intelhex
https://github.com/python-intelhex/intelhex/issues/45

closed 2020-10-08 by wojciechczyz