JelmerT/cc2538-bsl #103: New script name poll - please vote!

JelmerT, 2020-11-26:
See #41 and #89 
Time for a (non-binding) poll:

/polls Ti-BSL simple-bsl cc-bsl cc2538-bsl cc-Loader "Other (comment)"

JelmerT, 2020-11-26:
I like `simple-bsl` since the idea of the script is that it can support multiple BSLs in the future, not just Ti or CC ones. 
The code should also be pretty modular to be able to be expanded in this way.

omerk, 2020-11-26:
While having a name that will allow future expandability makes sense, realistically what are the chances of other devices being added? If the answer is "likely", then `simple-bsl` would do a good job as a generic name. If it is more of a "would be nice if it were to happen", my vote is on `cc-bsl` as it is what the script is, in its current state.

JelmerT, 2021-01-20:
The results are pretty clear! `cc-bsl` it is!

I will rename the repo and script and bump the version.

I'll try to warn and fix any tools/repos that use the script.

closed 2021-01-20 by JelmerT