JelmerT/cc2538-bsl #105: CC2652P support?

olili, 2020-12-23:
Thx for the great script. 

In my opinion/perception the CC2652P is not yet fully supported by your script. 
The chip is identified as CC1352P and "Verify" is not correctly implemented for it. 

Are there any plans to add this features? 

O.

JelmerT, 2021-01-20:
I currently don't have the bandwidth to implement/fix support for `CC2652P`, but I higly welcome PRs that add support!

If you can't add support yourself, can you describe what and where things are going wrong?

omerk, 2021-01-20:
CC2652 and CC1352 devices are pretty much identical (aside from radios), while chip ID needs some work programming and verification does work fine:

```
 $ ./cc2538-bsl.py -p /dev/ttyUSB0 -ewv CC1352P2_CC2652P_other_20210107.hex
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_other_20210107.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:81:19:36
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x710b8470)
```

@olili are you able to use BSL with Flash Programmer 2 on this board?

wilmardo, 2021-07-21:
I tried flashing on the `Ebyte E72-2G4M20S1E` today but I can't get it to pass the `(0x24)0x00028628` byte. It will then timeout:

```
# ./cc2538-bsl.py -Vewv CC1352P2_CC2652P_other_coordinator_20210120.hex
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_other_coordinator_20210120.hex
Your firmware looks like an Intel Hex file
For more solid firmware type auto-detection, install python-magic.
Please see the readme for more details.
Connecting to target...
*** sending synch sequence
Got 0 additional bytes before ACK/NACK
*** GetChipId command (0x28)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Version 0x3202F000
    Unrecognized chip ID. Trying CC13xx/CC26xx
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:81:58:63
    Performing mass erase
Erasing all main bank flash sectors
*** Bank Erase command(0x2C)
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Erase done
Writing 360448 bytes starting at address 0x00000000

<snip, more of the same>

Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x00028438
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x00028530
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x00028628
Traceback (most recent call last):
  File "./cc2538-bsl.py", line 1247, in <module>
    if cmd.writeMemory(conf['address'], firmware.bytes):
  File "./cc2538-bsl.py", line 668, in writeMemory
    self.cmdSendData(data[offs:offs+trsf_size])
  File "./cc2538-bsl.py", line 563, in cmdSendData
    if self._wait_for_ack("Send data (0x24)", 10):
  File "./cc2538-bsl.py", line 265, in _wait_for_ack
    raise CmdException("Timeout waiting for ACK/NACK after '%s'"
CmdException: Timeout waiting for ACK/NACK after 'Send data (0x24)'
ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'
```

It seems to connect just fine:
```console
# ./cc2538-bsl.py
Opening port /dev/ttyUSB0, baud 500000
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:81:58:63
```

@omerk Did it just work out of the box for you? Did you connect a BSL pin? I now just wire TX/RX and VCC/GND to my CP2102.

omerk, 2021-07-21:
@wilmardo I need to check but think I had to do a mass erase before I could program it, those modules might ship with BSL disabled.

wilmardo, 2021-07-27:
Little update, I can confirm that is works fine with the CC2652P. I hooked up the 3.3v on the PCB but that wasn't connected, it uses a converter from 5v. So when I connected 5v to the 5v pin everything went smooth.

For anyone stumbling on this issue in the future, you CAN read hardware info etc from just the RX/TX power. When flashing fails, make sure the chip is getting proper power.