JelmerT/cc2538-bsl #108: [Question] Cannot receive ACK after synchronization

dmitriy-serdyuk, 2021-01-06:
I am trying to flash a [device](https://makerspot.com/cc2640-bluetooth-low-energy-ble-usb-hid-dongle/) with cc2640R2F chip. This chip is connected to HID-to-UART interface CP2110.

I was able to establish a UART connection through HID using code based on [this](github.com/rginda/pycp2110). [Here](https://gist.github.com/dmitriy-serdyuk/8ec1db9d6d654a0391d105dbd54206bb) is a proxy serial interface I give instead of `self.sp`.

The example program loaded into CC2640 responds successfully:
``` python
>>> d.write(b'AT\r\n')
>>> while True:
...    print(d.read().decode('latin1'), end='')
AT
OK
<....> scanning
```

Unfortunately, it doesn't respond in the bootloader mode:
``` python
>>> cp2110_serial.setDTR(0)  # it looks that my bootloader pin is inverted
+DATM
>>> time.sleep(1)
>>> cp2110_serial.setRTS(0)  # go to sleep. reset pin also inverted??
>>> cp2110_serial.setRTS(1)  # wake up
>>> time.sleep(1)
>>> cp2110_serial.setDTR(1)
>>> cp2110_serial.write([0x55, 0x55])  # no response
```

What is interesting is that after I reset, it sends an ACK byte and then loads from flash
``` python
>>> cp2110_serial.setRTS(0)
>>> cp2110_serial.setRTS(1)
>>> while True:
...    r = d.read()
...    if not r:
...        continue
...    print(r, end='')
b'\x00\x33'
<....> # scanning
```

Any clues what is happening here? I appreciate any pointer.

JelmerT, 2021-01-20:
Do you have any schematics of the device?

Are you sure DTR and RTS pins are connected to the correct bootloader and reset pins on the cc2640R2F?

It looks like the bootloader is not being started. But the ack byte after the reset is interesting. Are you sure the bootloader is enabled and uses the correct pins in the firmware that's running on the cc2640R2F?

dmitriy-serdyuk, 2021-01-20:
Yeah, here [CC264 Dongle v02 Schematics.pdf](https://github.com/JelmerT/cc2538-bsl/files/5843417/CC264.Dongle.v02.Schematics.pdf). I am pretty sure that the pins are correct. According to the PDF, the reset pin is connected to GPIO.8 and the bootloader pin to GPIO.9. This corresponds to flicking a bit number [13 and 12](https://gist.github.com/dmitriy-serdyuk/8ec1db9d6d654a0391d105dbd54206bb#file-cp2110_serial-py-L18-L19) respectively.

> Are you sure the bootloader is enabled and uses the correct pins in the firmware that's running on the cc2640R2F?

I am not sure. Maybe they forgot to enable it at the factory? Is there a way to test this?