JelmerT/cc2538-bsl #110: Writing 360382 bytes starting at address 0x00000000 ERROR: Invalid data size: 360382. Size must be a multiple of 4.

sabas1080, 2021-04-28:
I am using code composer 10.2 and SDK 4.40

I have the next error with my file hex:

```
Writing 360382 bytes starting at address 0x00000000
ERROR: Invalid data size: 360382. Size must be a multiple of 4.
```

But in uniflash work fine

JelmerT, 2021-04-28:
This sounds like you're not using the correct file to flash, are you sure you're using the raw bin or hex file?

sabas1080, 2021-04-28:
Yes, I am using this hex 
[sniffer_fw.hex.zip](https://github.com/JelmerT/cc2538-bsl/files/6389192/sniffer_fw.hex.zip)

if I remove validation line, i have correct upload

JelmerT, 2021-04-28:
Which chip are you programming?

According to https://www.ti.com/lit/ug/swru333a/swru333a.pdf the download command needs a size that is a multiple of 4, and that's what the script is checking for.

sabas1080, 2021-04-28:
The chip is CC1352P1F3

sabas1080, 2021-04-30:
it seems the problem comes from the compiler

Issue related: https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/sub-1-ghz-forum/997796/cc1352p-file-hex-not-working-but-file-out-ok-with-example-sniffer/3686412#3686412