JelmerT/cc2538-bsl #111: SyntaxError: invalid syntax (Line 82)

Nornode, 2021-07-16:
Trying: 
**sudo python ./cc2538-bsl.py -p "/dev/ttyUSB1" -ewv "CC2652RB_coordinator_20210120.hex"**
results in:

```
File "./cc2538-bsl.py", line 82
    print(message, end=attr, file=sys.stderr)
                      ^
SyntaxError: invalid syntax
```

```
python --version
Python 3.9.6
```
And  Python 3.7.7

jayrobwilliams, 2022-03-07:
I had this same error with Python 3.5, 3.7, and 3.9 when using a conda environment. When I used pip3 to install the dependencies and used my system python to run the script, everything worked fine. Maybe there is a difference in one of the dependencies between pip and conda.

JelmerT, 2022-03-10:
This looks like an odd error, the error looks like you're running python 2, but when you do `--version` it is python 3.

Is there a reason why you're using `sudo`? and might that influence the python version that's being used? Could you try `sudo python --version`?

Is @jayrobwilliams his suggestion relevant to you?

jayrobwilliams, 2022-03-21:
I had to use `sudo` because I got a permissions error when I didn't. You're right that `sudo python` is invoking the system python instead of the conda one:
```
$ which python
/home/rob/miniconda3/envs/zigbee/bin/python
sudo which python
/usr/bin/python
```
Turns out this is due to `sudo` having a different PATH than a standard shell, and can be fixed by running `sudo conda activate <environment>` instead of simply `conda activate <environment>` https://stackoverflow.com/a/36659493.

JelmerT, 2022-03-22:
@Nornode did you see the same issue as @jayrobwilliams , did this fix your problem?