JelmerT/cc2538-bsl #113: Automatically start BSL mode

sshaikh, 2021-11-04:
I successfully used this script to flash an "ITead Sonoff Zigbee 3.0 USB Dongle Plus adapter based on Texas Instruments CC2652P", so thank you for the script.

There are two ways to put this model into BSL mode, pressing the button on the board (typically requiring assembly) or running a python script to do it via software.

I don't know if this is a feature of the stick, the UART bridge or the CC2652P, but I've attached the script for your perusal, as it may be useful for cc2538-bsl to do this when appropriate.

sshaikh, 2021-11-04:
[uartLog.txt](https://github.com/JelmerT/cc2538-bsl/files/7476867/uartLog.txt)

This is the Python script originally provided. It is designed to be run on Windows but can be made generic by deleting the COM suffix attached to the name when selecting the device.

JelmerT, 2021-11-04:
Thanks for that script @sshaikh . It looks like this is a simple toggle via the RTS and DTS lines.
This script should already be able to do this on its own, have you tried playing with these options?

>   --bootloader-active-high     Use active high signals to enter bootloader
>   --bootloader-invert-lines    Inverts the use of RTS and DTR to enter bootloader

sshaikh, 2021-11-04:
I have no idea how I missed those. I'll assume that works, unless you want a positive confirmation?

sshaikh, 2021-11-05:
FWIW neither of these options (or any combination of them) enabled BSL:

```
Opening port /dev/ttyUSB0, baud 500000
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

JelmerT, 2021-11-06:
Next step would be to check the schematic of the USB dongle and see how the RTS and DTR lines are connected to make sure they're toggled correctly. Maybe probe them with both scripts and see the difference.
Do you know if schematics are available somewhere?
Which serial-USB chip are they using?

sshaikh, 2021-11-07:
`lsusb` reports:

```
Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
```

More information here:

https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/

But shouldn't the script be explicit about what is required? If this is out of the scope of bsl, then I'm happy to use the peculiar uartlog script for my purposes.

JelmerT, 2021-11-08:
This is the code in their script that enters the bootloader:
```
def enterBoot(serialPort, delay=False):
    # delay is a workaround for bugs with the most common auto reset
    # circuit and Windows, if the EN pin on the dev board does not have
    # enough capacitance.
    # 如果开发板上EN管脚没有足够电容
    last_error = None

    # issue reset-to-bootloader:
    # RTS = either CH_PD/EN or nRESET (both active low = chip in reset
    # DTR = GPIO0 (active low = boot to flasher)
    #
    # DTR & RTS are active low signals,
    # ie True = pin @ 0V, False = pin @ VCC.
    setDTRState(serialPort, False)  # IO0=HIGH
    # setDTRState(serialPort, True)   # IO0=LOW
    setRTSState(serialPort, True)  # EN=LOW, chip in reset
    time.sleep(0.1)

    setDTRState(serialPort, True)  # IO0=LOW
    setRTSState(serialPort, False)  # EN=HIGH, chip out of reset

    time.sleep(1)

    setDTRState(serialPort, False)  # IO0=HIGH, done
    setRTSState(serialPort, False)
    time.sleep(1)

    return last_error
```
So it looks like both pins are active low, and RTS is connected to reset and DTR is connected to the bootloader pin. Which should be the same as in the standard configuration in this script.
So the only difference I'm seeing is the extra delays in their script. 
Try adding some delays in the same spots, and see if that makes it work.

sshaikh, 2021-11-08:
Just in case I've done it incorrectly, I was just adding the `--bootloader-active-high parameter` to the normal flashing command.

JelmerT, 2021-11-08:
> Just in case I've done it incorrectly, I was just adding the `--bootloader-active-high` parameter to the normal flashing command.

yes that's the correct way, but it looks like you shouldn't need it. (but still try it, together with `--bootloader-invert-lines`)

I'm guessing the delays are the issue here. You can verify with a scope on the reset and bootloader lines, or just try it with some trial and error.

Hedda, 2021-11-09:
Nice with option to automatically try both high and low signals to make adapter enter bootloader when Auto BSL is supported!

Auto BSL will also be supported by upcoming zzhp and zzhp-lite adapters by Electrolama:

https://electrolama.com/radio-docs/bsl/#auto-bsl

I have also confirmed that running the attached `uartLog.py` script does indeed make ITead's Sonoff Zigbee 3.0 Plus adapter enter bootloader mode and after that I could flash it directly with  `python cc2538-bsl.py -p COM5 -evw CC1352P2_CC2652P_launchpad_coordinator_20210708.hex` (also tested with electrolama fork using the commans `llama-bsl.py -p COM5 -b 115200 -evw CC1352P2_CC2652P_launchpad_coordinator_20210708.hex`) without having open its enclosure and pressing the BRL button which I think is very convenient. 

Note that as a bonus the `uartLog.py` script also lists all active COM ports on MS Windows (tested on Windows 10) so all you have you enter the number of the COM port, which again I find very convenient, but noted that the script is hardcoded for Windows COM ports.

As mentioned `uartLog.py` script attached in Sonoff doc look to manage to automatically enter BSL mode without the user having to manually enter whether the script should use active high signals to enter bootloader or inverts the use of RTS and DTR to enter bootloader (i.e. both active low ), and the ITead's Sonoff Zigbee 3.0 Plus adapter look to use both as activce low.

https://sonoff.tech/wp-content/uploads/2021/09/Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.docx

JelmerT, 2021-11-09:
@Hedda we're trying to not have to rely on `uartlog.py` though. Can you confirm that `cc-2538.py` can start the bootloader?

Hedda, 2021-11-10:
> @Hedda we're trying to not have to rely on `uartlog.py` though. 

I know, only wanted to let you know that I can confirm that the auto BSL feature is supported by the hardware and preflashed firmware, even though it is currently not working in the `cc-2538.py` script.

Can you confirm that `cc-2538.py` can start the bootloader?

> No, however, I can confirm that `cc-2538.py` can currently not start the bootloader.

Hedda, 2021-11-10:
> Which serial-USB chip are they using?

I can confirm that this Sonoff USB Plus dongle uses Silicon Labs CP2102N (USBXpress Family) according to ITead's specifications.

https://www.silabs.com/documents/public/data-sheets/cp2102n-datasheet.pdf

So need to use CP210x USB to UART Bridge VCP Drivers if using Windows OS:

https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

Noted there that ITead themselves are referring to the wrong device drivers for it on their own page, reviews and docs:

https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/

https://sonoff.tech/product-review/zigbee-3-0-usb-dongle-plus/

https://sonoff.tech/wp-content/uploads/2021/09/Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.docx

Hedda, 2021-11-10:
Probably not related but noted there are some errata posted for CP2102N which lists bugs and workarounds:

https://www.silabs.com/documents/public/errata/cp2102n-errata.pdf

Hedda, 2021-11-10:
ESP32/ESP8266 adapters with CP2102N A01 revision chips seem to have had similar issues on Windows OS. Some reports for those ESP devices say issue is fixed in hardware on CP2102N A02 revision chips and latest VCP drivers for Windows.

For CP2102N A01 revision chips they worked arroind the issues by adding more and longer delays as JelmerT suggest. 

Also, looks like Sonoff might have gotten part of boot mode code logic for uartLog.py script from esptool and esp_rfc2217_server:

https://github.com/espressif/esptool/blob/master/esp_rfc2217_server.py#L70

https://github.com/espressif/esptool/blob/master/esptool.py#L521

https://github.com/espressif/esptool/issues/136

https://github.com/espressif/esptool/pull/290

https://github.com/espressif/esptool/issues/399

https://github.com/espressif/esp-idf/issues/3925

https://www.esp32.com/viewtopic.php?t=11913

Hedda, 2021-11-16:
@guozi7788 Any chance you could report this issue to relevant personnel at ITead/Sonoff and get them to submit a fix in a patch?

Summery, "Auto BSL" is currently not working in the c2538-bsl script on Windows OS (Windows 10) with the SONOFF Zigbee 3.0 USB Dongle Plus by ITead even though its hardware does support "Auto BSL", this is likely due to the CP2102N USB-to-UART chip would need some additional delays added to the script, probably because of bugs in hardware or some issue Microsoft serial interface in Windows.

Please inform ITead/Sonoff that c2538-bsl is the most popular firmware upgrade script used by the community for flashing Texas Instruments based Zigbee USB dongles and having "Auto-BSL" (automatic reset and boot to bootloader via software) fully supporting SONOFF Zigbee 3.0 USB Dongle Plus by ITead would make it possible for the script to automatically enter BTL boot mode and perform the firmware upgrade without the need for to open the enclosure which would make the whole firmware upgrade procedure more convenient and user-friendly.

That is, having "Auto-BSL" being support in c2538-bsl for the SONOFF Zigbee 3.0 USB Dongle Plus by ITead would make the "uartLog.py" script recommended in ITead/Sonoff's existing firmware upgrade instructions obsolete and make it so much simpler for new users and beginners to upgrade the firmware using this c2538-bsl script as an easier to use tool. Also note that the c2538-bsl is a cross-platform script so it should work on all operating systems.

https://sonoff.tech/wp-content/uploads/2021/09/Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.docx

The current instructions from ITead/Sonoff suggest using the "uartLog.py" script attached in that Word document to enter the bootloader and then use TI's Flash Programmer 2 to flash the firmware, both of which are today only compatible with Microsoft Windows operating system, and that combination is not very convenient or easy to use, so not user-friendly at all.

guozi7788, 2021-11-16:
@Hedda
thanks for your advice.

uartLog.py is only used as a production testing tool to verify whether the dongle enters the boot mode, and it has not been given more uses.

As far as I know, the product manager is evaluating your needs，and you can check with your previous contacts for relevant progress.

Hedda, 2021-11-16:
> uartLog.py is only used as a production testing tool to verify whether the dongle enters the boot mode, and it has not been given more uses.

Actually, uartLog.py as a zip file with it is attached to https://sonoff.tech/wp-content/uploads/2021/09/Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.docx which is linked to from https://sonoff.tech/product-review/zigbee-3-0-usb-dongle-plus/ so FYI, anyone who bought the Sonoff Plus Dongle who want to upgrade the firmware is currently likely to follow that document and many will thus probably use that uartLog.py script to reset to boot mode rather than having to use a screwdriver to open the enclosure of the dongle, ...and as read it myself too I also posted a quick and dirty guide on how to achieve the wanted result using uartLog.py here at https://community.home-assistant.io/t/itead-sonoff-zigbee-3-0-usb-dongle-plus-adapter-based-on-texas-instruments-cc2652p/340705/105 as well as a copy of that guide here https://github.com/Koenkk/zigbee2mqtt/discussions/8840#discussioncomment-1631014_

JelmerT, 2021-11-16:
Could you try the version in this branch: https://github.com/JelmerT/cc2538-bsl/tree/feature/ITead_Sonoff_Zigbee-delay

Try it:
- without options
- with `--bootloader-invert-lines`
- with `--bootloader-active-high`
- with both `--bootloader-active-high --bootloader-invert-lines`

if none of those work, try increasing the delay https://github.com/JelmerT/cc2538-bsl/blob/efc94127624e08fce914ea88f56f7c0ca7e73041/cc2538-bsl.py#L239 all the way op to 1

guozi7788, 2021-11-17:
@Hedda 
I made a temporary modification and added a ‘sonoff’ parameter to automatically enter boot.

**python cc2538-bsl.py -p COM19 -evw CC1352P2_CC2652P_launchpad_coordinator_20210708.bin sonoff**

I tested it on windows and it looked like it worked correctly, you can try it.

[https://github.com/guozi7788/cc2538-bsl/blob/master/cc2538-bsl.py](url)

Hedda, 2021-11-17:
> Could you try the version in this branch: https://github.com/JelmerT/cc2538-bsl/tree/feature/ITead_Sonoff_Zigbee-delay
>
> Try it:
>
> * without options
> * with `--bootloader-invert-lines`
> * with [...]

@JelmerT Have now tested each of those options on my Windows 10 computer increasing the delay in that `time.sleep(0.1)` on line 239 testing 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 08., 0.9, and 1 but unfortunatly no better result. Keep getting the same timeout error:

 `ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)' `

Hedda, 2021-11-17:
> @Hedda I made a temporary modification and added a ‘sonoff’ parameter to automatically enter boot.
>
> **python cc2538-bsl.py -p COM19 -evw CC1352P2_CC2652P_launchpad_coordinator_20210708.bin sonoff** [...]

@guozi7788 tested and https://github.com/guozi7788/cc2538-bsl/blob/master/cc2538-bsl.py it works with the "sonoff" argument.

https://github.com/guozi7788/cc2538-bsl/commit/9f9060bcfcc0671e55934093aa1f952adf212123

Looks more complicated than adding some delays but that does indeed work for me with Sonoff Plus Dongle on Windows 10 OS.

dm82m, 2021-11-17:
also working on mac here:

```
(sonoff) ➜  sonoff python cc2538-bsl.py -p /dev/tty.usbserial-1420 -evw CC1352P2_CC2652P_launchpad_coordinator_20210708.hex sonoff
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
-----------ready to update-------------
Opening port /dev/tty.usbserial-1420, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C2:BA:FF
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x8c73378a)
(sonoff) ➜  sonoff
```

JelmerT, 2021-11-17:
@guozi7788 that's a great temp fix, but I can't pull that into the script of course. It's basically just the 2 separate scripts in the same file.

I don't have the hardware, so I can't look further into this. If anyone is willing to send me the HW I should be able to connect it to a logic analyzer or scope and see what's going on.

I more than welcome any PRs that integrate the functionality into the script of course.

Hedda, 2021-11-18:
> @guozi7788 that's a great temp fix, but I can't pull that into the script of course. It's basically just the 2 separate scripts in the same file.

That is fully understandable. Hope that ITead/Sonoff can ship you a free dongle so that you can take a deeper look at this!

Hedda, 2021-11-18:
FYI, according to comment in this discussion on the is a modified version of BSL Scripter that also got BSL working with CP2102:

https://e2e.ti.com/support/microcontrollers/msp-low-power-microcontrollers-group/msp430/f/msp-low-power-microcontroller-forum/802610/msp430f6736-msp430f6xxx-uart-bootloader-via-usb-bridge-ic?ReplyFilter=Answers&ReplySortBy=Answers&ReplySortOrder=Descending

That discussion also links to these other discussions which goes into much deeper details about how that solution works exactly:

https://e2e.ti.com/support/microcontrollers/msp-low-power-microcontrollers-group/msp430/f/msp-low-power-microcontroller-forum/734713/using-a-cp2102-with-bsl-scripter---a-solution

https://e2e.ti.com/support/microcontrollers/msp-low-power-microcontrollers-group/msp430/f/msp-low-power-microcontroller-forum/723144/msp430fr5969-bsl-configuration-using-cp2102

So apparently this is another type of needed invoke pattern is well known by other developers using CP2102N /  CP2102

Hedda, 2021-11-19:
> I don't have the hardware, so I can't look further into this. If anyone is willing to send me the HW I should be able to connect it to a logic analyzer or scope and see what's going on.
> 
> I more than welcome any PRs that integrate the functionality into the script of course.

@chiakikato read in https://github.com/Koenkk/zigbee2mqtt/discussions/8840 you have access to Sonoff Zigbee 3.0 USB Dongle Plus and already done some analyzing of it, any chance you could analyze further to find exact invocation sequence and pattern needed to invoke "Auto BLS"?

JelmerT, 2021-11-26:
I got my hands on a unit and have implemented the correct sequence to start the bootloader automatically.

The new option is now `--bootloader-sonoff-usb`, and you can find the PR here #114
Please test and add any issues to the PR. 

Thanks everyone!

Hedda, 2021-11-26:
> The new option is now `--bootloader-sonoff-usb`, and you can find the PR here #114

Nice! Will try to get time to test this weekend. If and when we know it is working will the "sonoff" parameter still be required?

Could otherwise CP2102 UART-to-USB chip or just USB "description" be detected and automatically use the correct sequence?

JelmerT, 2021-11-26:
the needed sequence between the "standard" connection (RTS and DTR connected 1 to 1 to reset and the bootloader pin) and the connection in the sonoff stick is quite different. So I don't think we could do them both automatically, I'd need to test with some different hardware though. If we could that'd be great, since then it would be transparent for all hardware.
Another option is to try 1 sequence, check if the chip responds, then try the other one. But this would add quite some delay while programming, because you're waiting for that timeout first.

Just detecting the CP2102 would not be a good idea, since many boards use that chip, or could use it in the future, and might not be connected in the same way to the radio being programmed. 
Itead also is still using the standard VID and PID for the CP2102, so we can't detect on that either.

Hedda, 2021-11-29:
> I got my hands on a unit and have implemented the correct sequence to start the bootloader automatically.
> 
> The new option is now `--bootloader-sonoff-usb`, and you can find the PR here #114 Please test and add any issues to the PR.

I can now confirm that with PR here #114 with `--bootloader-sonoff-usb` parameter is working for me on Windows 10. Thanks!

`cc2538-bsl.py -p COM5 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20210708.hex`

```
C:\Temp\cc2538-bsl-master-sonoff>cc2538-bsl.py -p COM5 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
sonoff
Opening port COM5, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C2:77:03
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x8c73378a)

C:\Temp\cc2538-bsl-master-sonoff>
```

Hedda, 2021-11-29:
> Just detecting the CP2102 would not be a good idea, since many boards use that chip, or could use it in the future, and might not be connected in the same way to the radio being programmed.
**Itead also is still using the standard VID and PID for the CP2102, so we can't detect on that either.**

FYI, reported before to ITead that they are not only using standard VID and PID for CP2102, they are also using the default "product description" from Silabs, and got a reply that will change the Product Description String value field to “Sonoff Zigbee 3.0 USB Dongle Plus” in their next batch in order to make it unique so that applications can use "product description" to automatically detected it. USB discovery in Home Assistant can already differ from different Zigbee adapters based solely on "product description" in USB chip EEPROM, (they will, however, they will continue standard VID and PID as changing those will require custom drivers). More info about that here -> https://community.home-assistant.io/t/community-help-wanted-to-whitelist-all-compatible-zigbee-and-z-wave-usb-adapters-for-automatic-discovery-in-home-assistant-os/344412

There are alread a few "product description" for CC2652 adapters in https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json

```
   {"vid":"10C4","pid":"EA60","description":"*2652*","known_devices":["slae.sh cc2652rb stick"]},
   {"vid":"10C4","pid":"EA60","description":"*tubeszb*","known_devices":["TubesZB Coordinator"]},
   {"vid":"1A86","pid":"7523","description":"*tubeszb*","known_devices":["TubesZB Coordinator"]},
   {"vid":"1A86","pid":"7523","description":"*zigstar*","known_devices":["ZigStar Coordinators"]},
```

Hedda, 2021-12-02:
> Got a reply from them that they plan on changing the Product Description String value field to “_Sonoff Zigbee 3.0 USB Dongle Plus_” on EEPROM in CP2102N for their next batch in order to at least make "product description" unique so that applications can use "product description" to automatically detected it.

FYI, ITead looks to have their Sonoff branded CC2652P based USB dongles back in stock again for $10.99 (US) at least for now:

https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/

No changelog announcement yet to let us know whether the USB chip in this batch has a unique  Product Description String.

Hedda, 2021-12-13:
FYI, ITead now posted a program (blob) that should let users themselves update the product description string on already shipped Sonoff dongles to supposedly match the same description already written to CP2102N EEPROM the latest batch of their dongles:

https://www.facebook.com/SONOFF.official/posts/2927251720919807

Tested that program myself on a Windows 10 computer and it did not work for me (program just closed without any messages).

Sad that ITead did not instead choose to release a script tool based on this open-source cp210x-program by VCTLabs:

https://github.com/VCTLabs/cp210x-program

http://cp210x-program.sourceforge.net/

https://sourceforge.net/projects/cp210x-program/

The point of having a unique custom product description in USB-to-UART chip is to allow USB auto-discovery, like the example in:

https://community.home-assistant.io/t/community-help-wanted-to-whitelist-all-compatible-zigbee-and-z-wave-usb-adapters-for-automatic-discovery-in-home-assistant-os/344412

tboy72, 2022-01-01:
Hello!
I also messed up the SONOFF USB ZIGBEE after router firmware the flash.
Unfortunately, flash again doesn't work at all.

`D:\Utils\cc2538-bsl-feature-ITead_Sonoff_Zigbee-delay>cc2538-bsl.py -p COM6 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_other_coordinator_20210708.hex
sonoff
Opening port COM6, baud 500000
Reading data from CC1352P2_CC2652P_other_coordinator_20210708.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'`

Is there another solution to the problem, or has USB become completely brick?
Thanks!

ZetaPhoenix, 2022-01-04:
FYI, I just got new adapters and they show up as `Sonoff Zigbee 3.0 USB Dongle Plus`

Hedda, 2022-01-07:
> I also messed up the SONOFF USB ZIGBEE after router firmware the flash. Unfortunately, flash again doesn't work at all.
>
>
> Is there another solution to the problem, or has USB become completely brick?

Yeah should have used the "`CC1352P2_CC2652P_launchpad_*.zip`" image file and not "`CC1352P2_CC2652P_other_*.zip`" image file.

[https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin)

Did you try removing enclosure and manually hold physical boot-button while plugging into USB-port?

[https://sonoff.tech/sonoff-zigbee-3-0-usb-dongle-plus-firmware-flashing-2/](https://sonoff.tech/sonoff-zigbee-3-0-usb-dongle-plus-firmware-flashing-2/)

If you messed up the bootloader firmware then believe only way to unbrick hardware is to flash with cJTAG, see details on that here:

[https://electrolama.com/radio-docs/advanced/flash-jtag/](https://electrolama.com/radio-docs/advanced/flash-jtag/)

The debug adapter for CC2652/CC1352 needs to support 2-pin cJTAG (a.k.a C-JTAG or c/JTAG) and not just common JTAG SWD as TI CC2652 debug does not use SWD and SWDCLK, etc. but instead only TMS and TCK (cJTAG is not SWD/SWDCLK). 

Texas Instruments XDS110 JTAG debug probe is a bit expensive but cheap clones of it should work, ex:

[https://aliexpress.com/item/4000751867419.html](https://aliexpress.com/item/4000751867419.html)

The official TMDSEMU110-U XDS110 JTAG Debug Probe from TI also looks to be out-of-stock now too.

[https://www.ti.com/tool/TMDSEMU110-U](https://www.ti.com/tool/TMDSEMU110-U)

Looking at pictures on [ITead's product page](https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/) you see that no SWD and SWDCLK pads are exposed.

So will probably need cJTAG debug probe with TMS and TCK like Texas Instruments XDS110 or clone.

Hedda, 2022-01-07:
> FYI, I just got new adapters and they show up as `Sonoff Zigbee 3.0 USB Dongle Plus`

Yeah some partial progress at least. Unfortunately, the script ITead posted that should let users themselves update the product description string on CP2102N for Sonoff dongles already shipped before does not seem to work:

https://www.facebook.com/SONOFF.official/posts/2927251720919807

Again sad that ITead did not choose to release a CP2102N write tool based on this open-source cp210x-program by VCTLabs:

https://github.com/VCTLabs/cp210x-program

That way both new and existing owners of it could utilize an automatic discovery feature if it is added.

Hedda, 2022-02-15:
Tip! Broken down the core steps to flash firmware mentioned in https://github.com/JelmerT/cc2538-bsl/pull/114 into a basic guide for flashing ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter on Windows OS but other than Serial/COM-port addressing the steps should be very similar on any other OS:

1. Install [Silabs CP210x device drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) (CP2102N USB-to-UART bridge / USB-to-Serial converter chip) needed for Windows OS.
2. Install [Python](https://www.python.org/downloads/) (in my case I installed and tested with Python for Windows 3.10.1).
3. Launch command-prompt (cmd.exe) as elevated Administrator. Upgrade pip with `python -m pip install --upgrade pip`, (if pip is not available then first run `python -m ensurepip --upgrade`), then pre-install dependencies fro cc2538-bsl from PyPi via pip command: `pip install wheel pyserial intelhex`
3. Optional but recommended: Download and install [zigpy-znp](https://github.com/zigpy/zigpy-znp) via pip/PyPI with `pip install zigpy-znp` then perform NVRAM backup by following instructions in [TOOLS.md](https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#backup-and-restore) (also find more details ZNP radio backup procedure at https://github.com/zigpy/zigpy/wiki/Coordinator-Backup-and-Migration) for Windows backup command should be something like `python -m zigpy_znp.tools.nvram_read COM5 -o nvram_backup.json` and optionally also backup Zigbee network via `python -m zigpy_znp.tools.network_backup COM5 -o network_backup.json`.
4. Download [cc2538-bsl on GiHub](https://github.com/JelmerT/cc2538-bsl/) from its ["master" branch](https://github.com/JelmerT/cc2538-bsl/archive/refs/heads/master.zip), (and not via pip/PyPI), unpact the zip to a folder then launch command-promt (cmd.exe) as elevated Administrator and run its `setup.py` to install its dependencies (should include "setuptools", "wheel", "pyserial", "intelhex" packages).
5. Get latest [firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin) (I tested latest "`CC1352P2_CC2652P_launchpad_*.zip`" image available from its ["master" branch](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin) at this time. Alternatively, you could get the latest "beta" version from the [develop branch](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)). Regardless make sure to get the correct image file as should be the one for "launchpad" (and not for "other)"! 
6. Stop any applications or services that might be connected to the Zigbee adapter via serial port (ex. Home Assistant's ZHA integration, Zigbee2MQTT, ioBroker, Jeedom, etc.). In my case, I ran the update on another computer so nothing to stop there.
7. Run command to flash from command-promt (cmd.exe) launched as elevated Administrator, example with firmware release available at this time: `python cc2538-bsl.py -p COM5 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20211217.hex`

closed 2022-02-26 by JelmerT

Hedda, 2022-02-28:
> > > FYI, I just got new adapters and they show up as `Sonoff Zigbee 3.0 USB Dongle Plus`
> > 
> > 
> > Yeah some partial progress at least. Unfortunately, the script ITead posted that should let users themselves update the product description string on CP2102N for Sonoff dongles already shipped before does not seem to work:
> 
> I tried making the script work on macOS and also tried the alternative tools, but had no success updating the descriptor or vid pid on mine. Manually selecting the port is not the worst, but it's unfortunate we can't make it happen automagically for those first units that were sent out with the standard descriptors.

@JelmerT Maybe you could look into VCTLabs cp210x-program to change the CP2102N "product description" string? See:

https://github.com/VCTLabs/cp210x-program

Note! Do not change the VID or PID as then will new a new device driver too, so only change the "product description" string.

PS: I will create a new separate issue with a request to auto-detect known dongles based on information provided on USB chip.