JelmerT/cc2538-bsl #116: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

Rowleys, 2021-12-26:
Hi, Im trying to do the python flash with the SSH, I'm doing it on the Home Assistant OS SSH,

Im getting this error  ..

```
/zigbee python3 cc2538-bsl.py -p /dev/ttyUSB0 -evw CC1352P2_CC2652P_launchpad_coordinator_20211217.hex
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20211217.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)
```

Thanks

omerk, 2022-01-03:
Your radio device is not in BSL mode (i.e: bootloader not running), follow the instructions for your particular radio device and try again.

Alternatively: If you have multiple USB devices plugged in, you might be using the wrong serial port. Make sure your radio device is indeed enumerating as `/dev/ttyUSB0` or (temporarily) unplug other USB devices and try again.

jeroenvdd, 2022-01-05:
its probably the bootloader, still need to figure out how to enable that option :)

but slae.sh website mentioned "

> You dont have to put the CC2652RB manually into the bootloader mode
> Its done by the tool automatically!
> So no need to reopen the enclosure for flashing..

in their how to so that's probably the reason why people aren't doing it :)

Hedda, 2022-01-10:
@Rowleys Check out https://github.com/JelmerT/cc2538-bsl/pull/114 (and [ITead_Sonoff_Zigbee-delay](https://github.com/JelmerT/cc2538-bsl/tree/feature/ITead_Sonoff_Zigbee-delay) branch) if you have ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter.

Rowleys, 2022-01-10:
I did yet it working. I had to add --bootloader-sonoff-usb to the command line

when I added this it flashed perfect

Mosher23, 2022-01-10:
> I did yet it working. I had to add --bootloader-sonoff-usb to the command line
> 
> when I added this it flashed perfect
> 
> sudo python3 cc2538-bsl.py -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb ../CC1352P2_CC2652P_launchpad_coordinator_20211217.hex

hello i am having same issue
but when i add `--bootloader-sonoff-usb` i get `option bootloader-sonoff-usb is not recognized`

what am i doing wrong?

Hedda, 2022-01-11:
@Rowleys Sorry are you saying that you specificalkly do have an [ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter](https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/) and that you now managed to flashing it using the [ITead_Sonoff_Zigbee-delay](https://github.com/JelmerT/cc2538-bsl/tree/feature/ITead_Sonoff_Zigbee-delay) branch from PR https://github.com/JelmerT/cc2538-bsl/pull/114 or? If so please post result in -> https://github.com/JelmerT/cc2538-bsl/pull/114 

@Mosher23  Do you also specifically have the [ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter](https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/) or which exact other adapter do you have? If you specificly have ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter then download correct branch of cc2538-bsl:

1. Download the correct "ITead_Sonoff_Zigbee-delay" branch from [#114](https://github.com/JelmerT/cc2538-bsl/pull/114)? Get -> https://github.com/JelmerT/cc2538-bsl/tree/feature/ITead_Sonoff_Zigbee-delay (because note the patch is not yet in the default "master" branch of cc2538-bsl):

    * https://github.com/JelmerT/cc2538-bsl/archive/refs/heads/feature/ITead_Sonoff_Zigbee-delay.zip
 
2. Be sure to download the correct firmware image named "CC1352P2_CC2652P_launchpad_*.zip" (and not the other one) from:

    * https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

      Alternativ get the latest experimental firmware image release from the "develop" branch:

      * https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

PS: If you have an an [ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter](https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/)  then please also post your results to -> https://github.com/JelmerT/cc2538-bsl/pull/114

juan11perez, 2022-01-13:
@Hedda 
I had the same issue as @Rowleys. After adding --bootloader-sonoff-usb to the command, flashing completed successfully.
I've added the feedback to 114 per your request.

Hedda, 2022-02-15:
FYI, after updating several times and helping others with different setups I have broken down my tested steps to flash firmware mentioned from https://github.com/JelmerT/cc2538-bsl/pull/114 into this basic guide for flashing ITead SONOFF Zigbee 3.0 USB Dongle Plus adapter. 

This guide uses the cc2538-bsl tool from JelmerT's latest "master" branch on GitHub and could easily be made to work on all operating systems. I tested on Windows OS but others have done this on Linux and Mac OS by using different Serial/COM-port addressing as the steps should otherwise be the same.

1. Install [Silabs CP210x device drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) (CP2102N USB-to-UART bridge / USB-to-Serial converter chip) needed for Windows OS and Mac OS, (this also required a reboot of my operating-system).
2. Install [Python](https://www.python.org/downloads/) (in my case I installed and tested with Python for Windows 3.10.1).
3. Launch command-prompt (cmd.exe) as elevated Administrator. Upgrade pip with `python -m pip install --upgrade pip`, (if pip is not available then first run `python -m ensurepip --upgrade`), then pre-install dependencies fro cc2538-bsl from PyPi via pip command: `pip install wheel pyserial intelhex python-magic`
3. Optional but recommended: Download and install [zigpy-znp](https://github.com/zigpy/zigpy-znp) via pip/PyPI with `pip install zigpy-znp` then perform NVRAM backup by following instructions in [TOOLS.md](https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#backup-and-restore) (also find more details ZNP radio backup procedure at https://github.com/zigpy/zigpy/wiki/Coordinator-Backup-and-Migration) for Windows backup command should be something like `python -m zigpy_znp.tools.nvram_read COM5 -o nvram_backup.json` and optionally also backup Zigbee network via `python -m zigpy_znp.tools.network_backup COM5 -o network_backup.json`.
4. Download [cc2538-bsl on GiHub](https://github.com/JelmerT/cc2538-bsl/) from its ["master" branch](https://github.com/JelmerT/cc2538-bsl/archive/refs/heads/master.zip), (and not via pip/PyPI), unpact the zip to a folder then launch command-promt (cmd.exe) as elevated Administrator and run its `setup.py` to install its dependencies (should include "setuptools", "wheel", "pyserial", "intelhex", and "python-magic" packages).
5. Get latest [firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin) (I tested latest "`CC1352P2_CC2652P_launchpad_*.zip`" image available from its ["master" branch](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin) at this time. Alternatively, you could get the latest "beta" version from the [develop branch](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)). Regardless make sure to get the correct image file as should be the one for "launchpad" (and not for "other)"! 
6. Stop any applications or services that might be connected to the Zigbee adapter via serial port (ex. Home Assistant's ZHA integration, Zigbee2MQTT, ioBroker, Jeedom, etc.). In my case, I ran the update on another computer so nothing to stop there.
7. Run command to flash from command-promt (cmd.exe) launched as elevated Administrator, example with firmware release available at this time: `python cc2538-bsl.py -p COM5 -e -v -w --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20211217.hex`

Obviously need to replace number in COM5 with the port # actually shown used under ports in Device Manager on Windows.

Also, if using Linux or Mac OS instead of Microsoft Windows then the `COM#` serial device path after `-p` will be different when set the port to use, (like for example `/dev/ttyUSB0` or `/dev/ttyUSB1`) and might need different [Silabs CP210x device drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) (CP2102N USB-to-UART bridge / USB-to-Serial converter chip) for your operating system, on Linux might have to run Python with `sudo`, and read that many on Linux successfully run `sudo python cc2538-bsl.py -e -v -w --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20211217.hex` without setting serial device path with `-p` for port to use.

mteusb, 2022-08-19:
I'm still having this problem.

I've done everything and I can't solve it.

Hedda, 2022-08-22:
Be sure now that you have bought a CC2652P-based “ZBDongle-P” adapter and not an EFR32MG21-based “ZBDongle-E” adapter.

FYI, ITead just released a [Silicon Labs EFR32MG21 “ZBDongle-E” adapter, which unfortunately is also marked as "Sonoff Zigbee 3.0 USB Dongle Plus V2"](https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-mcu-now-sold-for-19-99/442695) and looks the same as the CC2652P based "Sonoff Zigbee 3.0 USB Dongle Plus". This new “ZBDongle-E” adapter is really a different product as it is an improved version of their old [ITead’s previous barebone Silabs EFR32MG21 adapter 13](https://community.home-assistant.io/t/itead-zigbee-3-0-usb-dongle-stick-adapter-based-on-silicon-labs-efr32mg21/271144) and it is to be side-by-side as an alternative to [ITead’s TI CC2652P based “Sonoff Zigbee 3.0 USB Dongle Plus”](https://community.home-assistant.io/t/iteads-sonoff-zigbee-3-0-usb-dongle-plus-based-on-texas-instruments-cc2652p-20dbm-radio-mcu-now-sold-for-14-99/340705) (which has now renamed to “ZBDongle-P”).

https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-mcu-now-sold-for-19-99/442695

vesus

https://community.home-assistant.io/t/iteads-sonoff-zigbee-3-0-usb-dongle-plus-based-on-texas-instruments-cc2652p-20dbm-radio-mcu-now-sold-for-14-99/340705

mteusb, 2022-08-27:
> Be sure now that you have bought a CC2652P-based “ZBDongle-P” adapter and not an EFR32MG21-based “ZBDongle-E” adapter.
>
> FYI, ITead just released a [Silicon Labs EFR32MG21 “ZBDongle-E” adapter, which [...]

mine is EFR32MG21.

how do i bootloader and flash it?

Hedda, 2022-08-28:
> mine is EFR32MG21.
> 
> how do i bootloader and flash it?

Offtopic here but please note that I collected links in ZBDongle-E discussion here -> https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-v2-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-mcu-now-sold-for-19-99/442695

tothgergely92, 2023-05-04:
> I'm still having this problem.
> 
> I've done everything and I can't solve it.

As @Hedda wrote earlier: "Note! Also always need to first stop any application/service/integration connected to the adapter before use cc2538-bsl command." -> for me, that was the problem, after stopped zigbee2mqtt and other services related to it, the flashing was successful immediately without any error message.

smpillai11, 2024-02-24:
I had the same issue and the following fixed the problem
On windows PC
1. download cc2538-bsl.py and CC1352P2_CC2652P_launchpad_coordinator_20230507.hex (ensure file is unzipped)
2. go to command prompt and ensure python is installed by running command python. Then navigate to the folder with the above downloaded files
3. run command pip install wheel pyserial intelhex python-magic
4. Connect device to USB port and enter bootload mode by - press and hold RESET button, press and hold Key1 and then release RESET
5. run command python cc2538-bsl.py -p COM6 -evw --bootloader-sonoff-usb .\CC1352P2_CC2652P_launchpad_coordinator_20230507.hex

In the above command replace COM port (use device manager in PC to find COM port where the device is connected) and hex file name if required