JelmerT/cc2538-bsl #120: Docker support

git-developer, 2022-02-09:
I'm a Zigbee2MQTT user and found their guide for [Flashing the firmware via cc2538-bsl](https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_via_cc2538-bsl.html). I prefer keeping my host system clean and dislike installing dependencies to flash a USB dongle. Thus, I wrote a `Dockerfile` and a a small shell script to optionally download a firmware and then run c2538-bsl as a one-liner, e.g.
```shell
$ docker run --device /dev/ttyUSB0 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20211217.zip ckware/ti-cc-tool -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb
```

Maybe it's useful for someone else, so I'd like to notify you as the maintainers of cc2538-bsl about it. The project is currently hosted as [ti-cc-tool](https://github.com/git-developer/ti-cc-tool). If you're interested, the code could be moved as a contribution to your project.

Thanks for maintaining cc2538-bsl!

jansink, 2022-02-12:
Thanks for this! When I run this with the latest dev firmware of Koenkk I get: "ERROR: Invalid data size: 184245. Size must be a multiple of 4."  Any idea?

git-developer, 2022-02-12:
Hm... maybe an incorrect URL that doesn't point to a firmware but an HTML page?

git-developer, 2022-02-12:
Make sure the URL starts with https​://github.com/Koenkk/Z-Stack-firmware/**raw**/ and not with /**blob**/.

ismarslomic, 2022-05-20:
@git-developer you saved my day! I agree with your approach, I do also like to keep my host clean for dev dependencies. I managed flashing mine Sonoff Zigbee 3.0 dongle with your docker image. I did just an small tweek to mount my USB device to container by using /serial/by-id path, like this:

```(bash)
docker run \
--device /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b834cefbc512ec11968621c7bd930c07-if00-port0:/dev/ttyUSB0 \
-e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220219.zip \
ckware/ti-cc-tool \
-ewv \
-p /dev/ttyUSB0 \
--bootloader-sonoff-usb
```

Thanks @JelmerT for maintaining this solutio. 👍 from me to include Docker as alternative solution in the README.

JelmerT, 2022-08-02:
@git-developer this is pretty cool!

Feel free to submit a PR to update the readme on this script to include link to https://github.com/git-developer/ti-cc-tool

closed 2023-09-14 by git-developer