JelmerT/cc2538-bsl #121: Am I doing it right??

rvt, 2022-02-20:
Hey All,

I am using the Sonoff Zigbee2 dongles and writing it like this:

```
./cc2538-bsl.py -ewv --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20211217.hex
sonoff
Opening port /dev/tty.usbserial-1460, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_coordinator_20211217.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C1:6E:4E
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xba5c19c5)
Press any key to continue...
```

Note 1: I might have run it once without the --bootloader-sonoff-usb...
Note 2: I can always correctly write it

But when I try to use it with zigbee2mqtt it keeps on failing with `Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyUSB0'`
Note1: My zigbee2mqtt user is in the dialout group
Note2: For experimentation, I also start zigbee2mqtt as a root user

`
ls -ial /dev/ttyUSB0
27434295 crw-rw---- 1 root dialout 188, 0 Feb 20 15:27 /dev/ttyUSB0
`
dmesg shows:
```
[1101267.113756] usb 3-1.1: new full-speed USB device number 10 using xhci-hcd
[1101267.215651] usb 3-1.1: New USB device found, idVendor=10c4, idProduct=ea60
[1101267.215674] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1101267.215693] usb 3-1.1: Product: Sonoff Zigbee 3.0 USB Dongle Plus
[1101267.215708] usb 3-1.1: Manufacturer: ITead
[1101267.215722] usb 3-1.1: SerialNumber: c46c2db25a29ec11aa37757840c9ce8d
[1101267.216487] cp210x 3-1.1:1.0: cp210x converter detected
[1101267.217498] usb 3-1.1: cp210x converter now attached to ttyUSB0
```

questions:
1) Is there a way for mee to connect directly to ttyUSB0, for example with the screen command to see if it's alive? I tried several options, but so far I could not get de baudrate right.. Soe weird characters (which usually indicate wrong baudate...)
2) Did I brick it without a bootloader? If so, how do I tests that?

I am kinda in a loss understanding if my device is bricked or that it's something else, for example zigbee2mqtt not understand `/dev/ttyUSB0` somehow

JelmerT, 2022-02-20:
It looks like this script (cc2538-bsl) worked correctly. If it was able to erase and flash, your bootloader worked fine, and then verified, so your firmware should also be flashed on the chip correctly. If you can do that a second time, your bootloader also wasnt locked in the process, so all good there.

I'm not sure why zigbee2mqtt can't open the serial port, your best bet would be to check here https://github.com/Koenkk/zigbee2mqtt

You can try opening a serial port with `screen /dev/tty.usbserial-A601EFER 115200` (looks like most firmwares are usually 115200 and sometimes 57600) Do look up first how to exit a `screen` session though.
I like opening serial sessions in a terminal with this command `python3 -m serial.tools.miniterm --filter=direct /dev/tty.usbserial-A4013EFI 115200` which gives you all the nice colors and is a little easier than `screen`.
But I'm not sure if the zigbee coordinator firmware would give you any actual readable ascii data when you open the serial port though.

rvt, 2022-02-20:
Hey Jelmer,

thanks for the info... In this case I think I am writing it correctly then... and will head over to teh zigbee2mqtt community to understand why it keeps on showing the permission denied.

closed 2022-02-20 by rvt

JelmerT, 2022-02-20:
Actually, I think zigbee2mqtt is using the wrong port. Looks like you're flashing via this one: `/dev/tty.usbserial-1460` but then you're trying to access this one via zigbee2mqtt `/dev/ttyUSB0`, which might not even exist

rvt, 2022-02-20:
Hey ,

I have been flashing it on my mac that was using `/dev/tty.usbserial-1460`... But I am running zigbee2mqtt on a odroid (arm) system running Debian that shoudl be using `/dev/ttyUSB0` You can see that in dmesg where it says : `cp210x converter now attached to ttyUSB0`