JelmerT/cc2538-bsl #123: ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'

SuperPauly, 2022-05-02:
I'm running 
`python3 cc2538-bsl.py -p /dev/ttyUSB0 -e -v -w --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20220219.hex -V
`

And I'm getting this output:
`
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C9:20:33
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'
`

I have tried different baud rates and I always get the same error. How can I fix this?

SuperPauly, 2022-05-02:
UPDATE:
I changed USB ports and now it works but I'm getting a different error:

> sonoff
> Opening port /dev/ttyUSB0, baud 500000
> Reading data from CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
> Firmware file: Intel Hex
> Connecting to target...
> CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
> Primary IEEE Address: 00:12:4B:00:24:C9:20:33
>     Performing mass erase
> Erasing all main bank flash sectors
>     Erase done
> Writing 360448 bytes starting at address 0x00000000
> ERROR: No response from target on status request. (Did you disable the bootloader?)

JelmerT, 2022-05-02:
seems like in both cases your chip isnt answering after the write cycle. Are you sure you're using a correct firmware for your chip, and you're not disabling the bootloader in the chip?

ledhed2222, 2022-08-03:
@SuperPauly I was having the same issue. Decided to stop a bunch of user services. Homebridge, running on the same pi, seemed to be the culprit. Perhaps it was trying to access the same serial port for some reason? no idea