JelmerT/cc2538-bsl #124: Executing cc2538-bsl require pyserial which I already have installed

ionultd, 2022-05-28:
I execute: `python cc2538-bsl.py -p COM20 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20211217.hex`
and it says:
cc2538-bsl.py requires the Python serial library
Please install it with:
   pip3 install pyserial

Already done:

PS E:\mydownloads\cc2538-bsl-master> pip3 install pyserial
Requirement already satisfied: pyserial in c:\users\shane\.platformio\penv\lib\site-packages (3.5)

I am on windows 10 with Python 3.10.4

Any suggestion?

JelmerT, 2022-05-29:
could you try
`python3 cc2538-bsl.py -p COM20 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20211217.hex`

See if that works?

justin-davis-easyjet, 2023-10-25:
Having the same issue.. I've tried two computers and even tried using a python virtual environment.

justin4480, 2023-10-25:
Experiencing the same issue.. I've tried on two different linux computers and even tried using a python virtual environment.

It's odd as if I start a python environment from the command line I can manually import serial.

justin4480, 2023-10-25:
Actually found the issue...

`pip install pyserial intelhex python-magic`
>> Successfully installed...

`python cc2538-bsl.py -p /dev/ttyUSB0 -evw --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20230507.hex`
>> [Errno 13] Permission denied: '/dev/ttyUSB0'

`**sudo** python cc2538-bsl.py -p /dev/ttyUSB0 -evw --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20230507.hex`
>> cc2538-bsl.py requires the Python serial library...

This issue as I understand it is that running **sudo python filename.py** is not using the same python environment as running **python filename.py** and the pip install  commands were installing to the latter.

I resolved by pip installing as root, i.e.

`sudo pip install pyserial intelhex python-magic`
>> Successfully installed...

`python cc2538-bsl.py -p /dev/ttyUSB0 -evw --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20230507.hex`
>> ... Verified

Although, I don't believe it good practice to pip install as root.