JelmerT/cc2538-bsl #126: Timeout waiting for ACK/NACK after 'Send data (0x24)'.

entropical, 2022-06-19:
I have a new SONOFF Zigbee 3.0 USB Dongle Plus.

I am running the following commandline:

`python3 cc2538-bsl.py -Vewv -p /dev/ttyUSB0 --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_router_20220125.hex`

It works for a 5-10 seconds, sending data, and then dies. It dies after successfully sending varying amounts of data:

Attempt 1:
```
*** Send Data (0x24)0x0000D710
ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'
```
Attempt 2:
```
*** Send Data (0x24)0x00003548
ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'
```

Attempt 3:
```
*** Send Data (0x24)0x00024A18
ERROR: Timeout waiting for ACK/NACK after 'Send data (0x24)'
```

The start of each attempt looks like:

```
(.env) pi@henrypi4:~/Dev/cc2538-bsl $ python3 cc2538-bsl.py -Vewv -p /dev/ttyUSB0 --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_router_20220125.hex
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_router_20220125.hex
Your firmware looks like an Intel Hex file
For more solid firmware type auto-detection, install python-magic.
Please see the readme for more details.
Connecting to target...
*** sending synch sequence
Got 0 additional bytes before ACK/NACK
*** GetChipId command (0x28)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Version 0x3202F000
    Unrecognized chip ID. Trying CC13xx/CC26xx
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C8:58:EE
    Performing mass erase
Erasing all main bank flash sectors
*** Bank Erase command(0x2C)
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Erase done
Writing 360448 bytes starting at address 0x00000000
*** Download command (0x21)
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Send Data (0x24)0x00000000
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
```

I tried lowering the baud rate, even as low as 9600, but it still fails.

Plugged into a Raspberry Pi4B, with official power supply. I also tried multiple ports on the Pi.

A message suggested installing python-magic, which I did, but it made no difference.

I sometimes get errors before it sends data, e.g. 

- ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
- ERROR: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

entropical, 2022-06-19:
I tried it on a different Pi, and it worked.

Maybe the USB ports on my Pi4 are b0rked.

JelmerT, 2022-08-02:
Glad it worked out, closing this issue

closed 2022-08-02 by JelmerT