JelmerT/cc2538-bsl #127: --ieee-address does not update work with sonoff zigbee 3.0 USB dongle plus

mafu75, 2022-07-25:
Hi all,

I am trying to alter the coordinator address of my above CC2652P based USB dongle like:

python3 cc2538-bsl.py --bootloader-sonoff-usb -evw --ieee-address 00:12:4b:00:01:4d:12:34 ../CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from ../CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C0:88:01
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xddfc152d)
Setting IEEE address to 00:12:4b:00:01:4d:94:3d
Writing 8 bytes starting at address 0x00057FC8
Write 8 bytes at 0x00057FC8
    Set address done

But I always get the same output. I would expect the newly written ieee address as new primary address when relaunching the script. Also zigbee2mqtt later still reports the old address.

So maybe  0x00057fc8 as destination for the new ieee address is incorrect?

Any hints?

JelmerT, 2022-07-26:
I'm pretty sure there's 2 ieee addresses in this chip. You're reading the first one, and setting the second one. I'm pretty sure the primary is hard coded and secondary is configurable.

JelmerT, 2022-07-26:
Why zigbee2mqtt isn't using the secondary address, I'm not sure about that one.

You could read out `0x00057FC8` to check your new address is actually there, to determine where the issue is.

We should probably print out both addresses, feel free to submit a PR on this feature.

gurues, 2022-08-29:
> Why zigbee2mqtt isn't using the secondary address, I'm not sure about that one.
> 
> You could read out `0x00057FC8` to check your new address is actually there, to determine where the issue is.
> 
> We should probably print out both addresses, feel free to submit a PR on this feature.

The same thing happens to me with the sonoff dongle and I can check with Flash Programmer that the secondary address is updated to the value I require. Do I now have to update the primary address? Will the device continue to work or can I charge it?
Thank you

jcat59, 2023-02-10:
I am using the below command to flash my sonoff CC2652P with router software. Is there a way to update this command line so that I can update the ieee address?

python cc2538-bsl.py -p COM3 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_router_20221102.hex

JBS5, 2024-06-11:
Any update on this? Was trying to prepare a Sonoff Zigbee USB Dongle Plus as a backup coordinator by copying the IEEE adres from my primary stick to the backup one, but seems to change the secondary IEEE where Z2M is using the primary.