JelmerT/cc2538-bsl #129: Which Pyhon version is supported?

samip5, 2022-08-11:
Hi,

I tried to find out which Python version is required as I'm having issues trying to use it with python3.10.

```
python3 ./cc2538-bsl.py -V -p socket://192.168.2.112:6638 -evw CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Traceback (most recent call last):
  File "/home/sky/Downloads/cc2538-bsl-2.1/venv/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: 'socket://192.168.2.112:6638'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sky/Downloads/cc2538-bsl-2.1/./cc2538-bsl.py", line 1042, in <module>
    cmd.open(conf['port'], conf['baud'])
  File "/home/sky/Downloads/cc2538-bsl-2.1/./cc2538-bsl.py", line 195, in open
    self.sp = serial.Serial(
  File "/home/sky/Downloads/cc2538-bsl-2.1/venv/lib/python3.10/site-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/home/sky/Downloads/cc2538-bsl-2.1/venv/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port socket://192.168.2.112:6638: [Errno 2] No such file or directory: 'socket://192.168.2.112:6638'
ERROR: [Errno 2] could not open port socket://192.168.2.112:6638: [Errno 2] No such file or directory: 'socket://192.168.2.112:6638'
```

JelmerT, 2022-08-11:
Python 3 should be supported.

Looks like the script cant find `socket://192.168.2.112:6638` is this a socket to a remote serial port? I've never used the script like this before, what's your hardware setup?

samip5, 2022-08-11:
> Python 3 should be supported.
> 
> Looks like the script cant find `socket://192.168.2.112:6638` is this a socket to a remote serial port? I've never used the script like this before, what's your hardware setup?

It's an remote serial port, yes. It consists of CC2652P2 Radio Module + Olimex ESP32-PoE Module. ESP32 is running ESPHome and it's configured in way that that port goes to the radio module.

Funny thing, it actually did find it when done inside Docker as I was wiresharking my nic when trying to run it and it in fact did not even try to connect to it but inside the Docker container, I was able to use it?

```
docker run --rm  -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220219.zip ckware/ti-cc-tool -ewv -p socket://192.168.2.112:6638
Unable to find image 'ckware/ti-cc-tool:latest' locally
latest: Pulling from ckware/ti-cc-tool
59bf1c3509f3: Already exists 
aa9c682af02e: Pull complete 
7908a6ef3409: Pull complete 
Digest: sha256:2fac6f51375481dee2edee84f866f8640b8cca650e193a860e0cdc4dc8c287ad
Status: Downloaded newer image for ckware/ti-cc-tool:latest
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220219.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20220219.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20220219.hex'
Opening port socket://192.168.2.112:6638, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: <snip address>
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xddfc152d)
```

JelmerT, 2022-08-11:
Can you manually open the serial port through that socket? with a serial terminal?

samip5, 2022-08-11:
zigbee2mqtt utilizes that socket as well without issues, and I also tried from a Ubunru system but faced the same issue so I'm not at all sure where the issue with it would be.

JelmerT, 2022-08-12:
Note that this script needs access to more pins than just TX and RX to get the hardware into bootloader mode. Please check if that is possible through the socket connection.

samip5, 2022-08-12:
> Note that this script needs access to more pins than just TX and RX to get the hardware into bootloader mode. Please check if that is possible through the socket connection.

It did not even try to connect according to Wireshark, so it has to be a Python version compatibility issue however I have already upgraded the firmware by using the Docker container so for me, this has been resolved by using a workaround.
The Docker container seems to be using the same script (https://github.com/git-developer/ti-cc-tool).

Just for reference:

OS: Gentoo & Ubuntu 22.04
Python: 3.10 and 3.8

Original issue on Tube's repo: https://github.com/tube0013/tube_gateways/issues/80

Both failed to even try to connect and rather said that not much telling error.

JelmerT, 2022-08-12:
The socket connection only works on POSIX systems, but looks like you're testing with one, so that shouldn't be the issue.

I'm indeed suspecting PySerial not being happy with something, not sure what though.

If you're up for more debugging, could you try connecting with miniterm to the socket and see if it works that way:
```
python3 -m serial.tools.miniterm --filter=direct socket://192.168.2.112:6638 115200
```

closed 2024-09-23 by JelmerT