JelmerT/cc2538-bsl #133: [REQUEST] SL Web Tools implementation of cc2538-bsl for flashing TI firmware via a web browser?

Hedda, 2023-02-10:
@JelmerT  Home Assistant founders have now among other things announced "[SL Web Tools](https://github.com/NabuCasa/sl-web-tools)" project (short for "Silicon Labs Web Tools") is available as open source, so wondering if you would consider forking that to make a project called "TI Web Tools" (short for "Texas Instruments Web Tools") or similar that is based on that in combination with [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl) code for the TI flashing part?

https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/#silabs-multi-flasher--sl-web-tools

(Note that this is only just one section from https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/ )

SL Web Tools allows users to manage the firmware on Zigbee/Thread sticks directly from a web browser, with no terminals or compilation necessary. This allows any creator to offer easy firmware updates to their users and make experimental firmware updates available to test bug fixes quickly.

SL Web Tools runs in the browser using [Pyodide](https://pyodide.org/en/stable/) and is currently powered by another new open-source Python project from them, “[silabs-universal-flasher](https://github.com/NabuCasa/universal-silabs-flasher)”, which is obviously for Silicon Labs based SoCs, so request/idea would be to replace that with [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl) in order to allow flashing of Texas Instruments based adapters instead, do you think that would be possible?

https://github.com/NabuCasa/sl-web-tools

(and currently powered by https://github.com/NabuCasa/universal-silabs-flasher for Silicon Labs based chips)

https://www.youtube.com/watch?v=-88K23e8XYw&t=2s&ab_channel=HomeAssistant

To see it in action, a demo of SL Web Tools for Silicon Labs chips has been added to the Home Assistant SkyConnect documentation.

https://skyconnect.home-assistant.io/

[image]

Note! [SL Web Tools](https://github.com/NabuCasa/sl-web-tools) is in turn a fork of their previous "[ESP Web Tools](https://esphome.github.io/esp-web-tools/)" project which allow DIY firmware flashing of ESP32/ESP8266 SoC projecst to offer an easy web installer on their homepage, allowing users to get started directly from their browser. It has been adopted by all the great projects, including [Tasmota](https://tasmota.github.io/install/), [WLED](https://install.wled.me/), and [ESPresense](https://espresense.com/firmware). Ever since the introduction of ESP Web Tools, they expanded this to more devices, and now they also introduced [SL Web Tools](https://github.com/NabuCasa/sl-web-tools) for SiLabs based chips.

PS: For reference; the firmware build system for Silicon Labs was made by Home Assistant developer/engineer and Nabu Casa employee [Stefan Agner (a.k.a. agners](https://github.com/agners), who also developed the EFR32MG21-based [Home Assistant SkyConnect USB stick/dongle](https://www.home-assistant.io/skyconnect) as well as the EFR32MG21-based radio module integrated into the Home Assistant Yellow appliance for Rasberry Pi 4 Compute Module). And the "universal-silabs-flasher" firmware tool was made by [puddly](https://github.com/puddly) who is also one of the lead developers of [zigpy ](https://github.com/zigpy/zigpy)and the [radio libraries for zigpy](https://github.com/zigpy) that is used in Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) (among other projects).

Hedda, 2023-03-29:
If either @JelmerT or some other developer is interested in the development of a Web Tools implementation for Texas Instruments CC2652 and CC1352 boards/adapters then you should know that [omerk](https://github.com/omerk) from [Electrolama](https://github.com/electrolama) (of [zzh](https://electrolama.com/projects/zig-a-zig-ah/) and [zoe/zoe2](https://electrolama.com/projects/zoe2/) fame) has offered to sponsor to anyone interested in implementing this with free Zigbee Coordinator hardware from [electrolama.com](https://electrolama.com/) as part of his/their community projects initiative though, see details here -> https://github.com/electrolama/community-projects

JelmerT, 2023-03-29:
Really cool, thanks for posting @Hedda !

I currently don't have the bandwidth to work on this, but if I can support in any other way I'd be happy to.

Hedda, 2023-05-19:
If there is anyone else with time, skill and interest to make this idea happen then also be sure to checkout thhis fork too:

https://github.com/darkxst/sl-test

darkxst have extended to with [manifests](https://github.com/darkxst/sl-test/tree/main/manifests) support unique per dongle settings (such as for example the recommended baud rate).

https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs/manifests

See sample [manifest](https://github.com/darkxst/sl-test/blob/gh-pages/assets/manifests/zbdongle-e.json) file:

https://github.com/darkxst/sl-test/blob/gh-pages/assets/manifests/zbdongle-e.json

Visit SL test website and the source code for the webpages on that site:

https://darkxst.github.io/sl-test/

https://github.com/darkxst/sl-test/tree/gh-pages

Hedda, 2023-05-19:
Also see this other related suggestion to copy the command syntax from the "Universal Silicon Labs Flasher" project -> https://github.com/JelmerT/cc2538-bsl/issues/136