JelmerT/cc2538-bsl #137: Timeouts are set too short for flashing over LAN.

Steltek, 2023-06-08:
Trying to update the firmware on a Zigstar PoE (Zigbee 2 LAN coordinator) and it kept failing with a timeout at various addresses:

> server:/swdownloads/cc2538-bsl/cc2538-bsl-master# python3 cc2538-bsl.py -p socket://zigstargw.local:6638 -b56000 -evw /Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
> Opening port socket://zigstargw.local:6638, baud 56000
> Reading data from /Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
> Firmware file: Intel Hex
> Connecting to target...
> CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
> Primary IEEE Address: 00:XX:XX:XX:XX:XX:XX:XX
>     Performing mass erase
> Erasing all main bank flash sectors
> ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'
> 
> 
> server:/swdownloads/cc2538-bsl/cc2538-bsl-master# python3 cc2538-bsl.py -p socket://zigstargw.local:6638  -evw /Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
> Opening port socket://zigstargw.local:6638, baud 500000
> Reading data from /Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
> Firmware file: Intel Hex
> Connecting to target...
> CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
> Primary IEEE Address: 00:XX:XX:XX:XX:XX:XX:XX
>     Performing mass erase
> Erasing all main bank flash sectors
>     Erase done
> Writing 360448 bytes starting at address 0x00000000
> ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'

I ended up going through the code and doubling (or tripling) all of the timeout values (from 2 to 4 or 8, ...) and it worked straight away after that:

> server:/swdownloads/cc2538-bsl/cc2538-bsl-master# python3 cc2538-bsl.py -p socket://zigstargw.local:6638  -evw /Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
> Opening port socket://zigstargw.local:6638, baud 500000
> Reading data from /Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
> Firmware file: Intel Hex
> Connecting to target...
> CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
> Primary IEEE Address: 00:XX:XX:XX:XX:XX:XX:XX
>     Performing mass erase
> Erasing all main bank flash sectors
>     Erase done
> Writing 360448 bytes starting at address 0x00000000
> Write 104 bytes at 0x00057F988
>     Write done
> Verifying by comparing CRC32 calculations.
>     Verified (match: 0xe83aa727)
> server:/swdownloads/cc2538-bsl/cc2538-bsl-master#
> 

Perhaps the timeouts could automatically be extended when a network connection is used instead of a local serial one?

Ulrar, 2024-10-02:
Hey, thanks for that, I had the same issue.
For anyone else stumbling upon this in the future a little tip, there's an unlabelled timeout on line 383, last argument to the _wait_for_ack call has 2 seconds which is just a tad too low for me.

JelmerT, 2024-10-02:
@Steltek @Ulrar thanks for the comments! feel free to submit a PR to fix this 👍

Steltek, 2024-10-03:
@JelmerT Done, though a little disclaimer: These are the values that worked for me but may not be optimal for everyone. I no longer have the ability to run tests (the device has since been migrated to different firmware).

PR #181 (2025-01-14): Raising timeouts for operation over LAN.
Fixes #137 by raising timeout values acceptable for flashing over LAN.
closed 2025-01-14 by JelmerT