JelmerT/cc2538-bsl #147: SyntaxError invalid character

RoyalTS83, 2023-11-02:
Hello,
can someone help me with this?

C:\>python cc2538-bsl.py -p /dev/ttyUSB0 -evw --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
  File "C:\cc2538-bsl.py", line 95
    <title>cc2538-bsl/cc2538-bsl.py at master · JelmerT/cc2538-bsl · GitHub</title>
                                              ^
SyntaxError: invalid character '·' (U+00B7)

JelmerT, 2023-11-06:
@RoyalTS83 it looks like you pasted this some where:
`<title>cc2538-bsl/cc2538-bsl.py at master · JelmerT/cc2538-bsl · GitHub</title>`
Maybe you copy pasted a little too much from a website somewhere, or you edited the script itself by accident?

closed 2024-09-23 by JelmerT

djurny, 2024-09-26:
Hi there,
This happens when you try to download [cc2538-bsl.py](https://github.com/JelmerT/cc2538-bsl/blob/master/cc2538-bsl.py) using the wrong method. The link github provides to you is an online version of the file, not a direct link to download the file.
To prevent this, use the online version of the file and there you press the download icon to download the actual file instead of an HTML page of the online version.