JelmerT/cc2538-bsl #160: [NEW HARDWARE] CC2674 (CC2674P/CC2674R) and CC1354 (CC1354P) compatibility?

Hedda, 2024-03-15:
Anyone tested compatibility with the new CC2674/CC2674P/CC2674R and/or CC1354/CC1354P chips from Texas Instruments?

https://www.ti.com/product/CC2674P10

https://www.ti.com/product/CC2674R10

https://www.ti.com/document-viewer/CC2674P10/datasheet

https://www.ti.com/document-viewer/CC2674R10/datasheet

https://www.ti.com/document-viewer/lit/html/swra765

https://www.ti.com/lit/gpn/CC2674P10

https://www.ti.com/lit/gpn/cc2674r10

SMLIGHT has announced/launched CC2674P (CC2674P10) based "SLZB-07p10" USB dongle and "SLZB-06p10" Ethernet/WiFi adapter:

SLZB-07p10 -> https://smlight.tech/product/slzb-07p10/

SLZB-06p10 -> https://smlight.tech/product/slzb-06p10/

CC2674P10 based E72-2G4M20S1C radio module has also been available from Ebyte/Cdebyte for a while now for DIY builds and more:

https://www.cdebyte.com/products/E72-2G4M20S1C

(at least that radio moduule is pin to pin compatible with their CC2652P module https://www.cdebyte.com/products/E72-2G4M20S1E)

I understand CC2674P10 should be able to be a replacement of CC2652P2/CC2652P7, coming in the same package size and format?

https://www.ti.com/wireless-connectivity/zigbee/products.html#1241=Wireless%20MCU&

https://www.ti.com/wireless-connectivity/zigbee/overview.html

I don't own any of these myself but devs can check out official "LP-EM-CC1354P10" (CC1354P10 LaunchPad) development kits hardware

* https://www.ti.com/product/CC1354P10#hardware-development
  * https://www.ti.com/tool/LP-EM-CC1354P10
    * https://www.ti.com/tool/LP-XDS110
  * https://www.ti.com/tool/LP-XDS110ET

It looks like "CC1354P10-1 LaunchPad" and "CC1354P10-6 LaunchPad" is supported in SimpleLink CC13xx CC26xx SDK version 7.10.01.24?

I've not yet seen any news about a CC2674P10 or CC2674R10 based LaunchPad, but Mouser and others now look to have SoCs in stock.

* CC2674P106T0RGZR (CC2674P10 SoC)
* CC2674R106T0RGZR (CC2674R10 SoC)

Anyway, CC2674P10 + CC1354P10 chips are readily available so guess won't be long before will also see such adapters from others too:

* [CC2674P10](https://www.ti.com/product/CC2674P10) = 2.4GHz radio SoC for Bluetooth 5.2 LE, Thread, Zigbee 3.0 with 48-MHz ARM Cortex-M33, 296KB SRAM, 1024KB Flash.

* [CC1354P10](https://www.ti.com/product/CC1354P10) = Multi-Band radio SoC for Wi-SUN, Bluetooth 5.2 LE, Thread, Zigbee 3.0 with 48-MHz ARM Cortex-M33, 288KB SRAM, 1024KB Flash.

PS: There is no community firmware from Koenkk yet but that is probably partially because no one made a CC2674 or CC1354 adapter yet, however Koenkk and others in the Zigbee2MQTT community are working on it, check out this discussion about community firmware build for CC26x4/CC13x4 -> https://github.com/Koenkk/Z-Stack-firmware/discussions/476

PPS: More referenses:

[image]

[image]

[image]

[image]

darkxst, 2024-04-04:
CC2674P10 has a completely new memory layout compared to previous chips.

SMLIGHT will be releasing a flasher with support for the p10 chips though.

kkrentz, 2024-08-05:
The SIMPLELINK-CC13XX-CC26XX-SDK 7.41.00.17 contains a modified version of cc2538-bsl with support for newer TI chips. I haven't tried it, yet.

Hedda, 2024-08-05:
> The SIMPLELINK-CC13XX-CC26XX-SDK 7.41.00.17 contains a modified version of cc2538-bsl with support for newer TI chips

Do you have link to the source code?

Hedda, 2024-08-05:
> > The SIMPLELINK-CC13XX-CC26XX-SDK 7.41.00.17 contains a modified version of cc2538-bsl with support for newer TI chips
> 
> Do you have link to the source code?

Here is the link for 7.40.00.77 update (from 5 months ago) but I could not find links to new 7.41.00.17 update on GitHub:

https://github.com/TexasInstruments/simplelink-lowpower-f2-sdk/tree/main/source/third_party/cc2538-bsl

Maybe the 7.41.00.17 update does not contain any updates to the third-party cc2538-bsl code?

kkrentz, 2024-08-05:
Indeed, 7.41.00.17 has the same version of cc2538-bsl as 7.40.00.77.

Hedda, 2024-08-05:
By the way, SMLIGHT is shipping their CC2674P1 based SLZB-07p10 USB adapter dongle and is now selling for just under $15(US):

https://smlight.tech/product/slzb-07p10/

Note! They updated that webpage to specifically state there that only for developers for now "_**ZIGBEE FIRMWARE FOR ZIGBEE CHIP CC2674P10 IS STILL UNDER DEVELOPMENT**_" and that "_**YOU CAN BUY THIS DEVICE ONLY FOR DEVELOPMENT PURPOSES**_".

> CC2674P10 GPIOS for developers (corresponds to LAUNCHXL board CC2652P):
> * RF: RF_24GHZ → DIO28, RF_HIGH_PA → DIO29;
> * UART: TX → DIO13, RX → DIO12;
> * LEDs: GREEN LED → DIO6, RED LED → DIO7;
> * BOOT: BSL → DIO15, Reset → RESET_N;

darkxst, 2024-08-13:
SMLIGHT have released a web flasher with support for P10 now. 
https://smlight.tech/flasher/#SLZB-07p10

It is based on a heavily modified and improved fork of this project (which also includes P10 support).
https://pypi.org/project/smlight-cc-flasher/

Hedda, 2024-08-13:
> SMLIGHT have released a web flasher with support for P10 now. https://smlight.tech/flasher/#SLZB-07p10
> 
> It is based on a heavily modified and improved fork of this project (which also includes P10 support). https://pypi.org/project/smlight-cc-flasher/

Cool, but do you also know if they host the code for thhose in public Git repositories on GitHub as well or?

darkxst, 2024-08-14:
https://github.com/smlight-tech/smlight-cc-flasher
https://github.com/smlight-tech/smlight-cc-web-tools