JelmerT/cc2538-bsl #161: -p/--port option seems broken

6wheels, 2024-03-15:
Hi guys,

I'm trying to flash my Sonoff dongle version P as usual, but it fails with this error:
```
cc2538-bsl.py: error: argument -p/--port: invalid int value: '/dev/ttyUSB1'
```

Here is the command I use (using docker):
```
docker run --rm --device /dev/ttyUSB1:/dev/ttyUSB1 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip ckware/ti-cc-tool -ewv -p /dev/ttyUSB1 --bootloader-sonoff-usb
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip
```

I double checked the port on which the Sonnoff dongle is connected.

After reading the commit history, I can see some changes have been made to use argparse (https://github.com/JelmerT/cc2538-bsl/commit/f42a2e252f27a6e489c3f76e63ec9543121c29dd#diff-9d6356e73eb9065e1b5a72ad6cfd71d5c08cd22b7be6cb08d6318aa51885a93eR1061)

The option expects an int value and port seems to be dynamically retrieved from `/dev/`.
So I tried several things, but it always fails.

Any help appreciated.
Thanks.

joshfedo, 2024-03-15:
I just ran into this as well. Use the number of the usb:
ie  ../../ttyUSB0 = -p 0
../../ttyUSB1 = -p 1

6wheels, 2024-03-15:
Hi @joshfedo, 
Thanks for answering, unfortunately, I already tested this, and I get another error, which is:
```
docker run --rm --device /dev/ttyUSB1:/dev/ttyUSB1 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip ckware/ti-cc-tool -ewv -p 1 --bootloader-sonoff-usb
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20230507.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20230507.hex'
ERROR: "port" must be None or a string, not <class 'int'>
```

So int or string? Make a choice! 😆

sendcard, 2024-03-16:
i have run this
python cc2538-bsl.py -p --com3 -e -v -w --bootloader-sonoff-usb 
CC1352P2_CC2652P_launchpad_coordinator_20230507.hex

Same error
usage: cc2538-bsl.py [-h] [-q] [-V] [-f] [-e] [-E ERASE_PAGE] [-w] [-W] [-v] [-r] [-l LEN] [-p PORT] [-b BAUD]
                     [-a ADDRESS] [-i IEEE_ADDRESS] [--bootloader-active-high] [--bootloader-invert-lines]
                     [--bootloader-sonoff-usb] [-D] [--version]
                     file
cc2538-bsl.py: error: argument -p/--port: expected one argument

0sy, 2024-03-16:
Same here:

cc2538-bsl.py: error: argument -p/--port: invalid int value: 'COM5'

Yogy42, 2024-03-16:
First time user:
Same problem 
`cc2538-bsl.py: error: argument -p/--port: invalid int value: '/dev/ttyUSB0'`

accordxtc, 2024-03-16:
Add me to the list for having the same issues. What are the odds I go to change my zigbee setup and now can't flash it because of this issue. Any idea on a fix?

git-developer, 2024-03-17:
You can use the previous image `ckware/ti-cc-tool:2024-01-28` as a workaround.

git-developer, 2024-03-17:
I think that the change introducing this behavior in https://github.com/JelmerT/cc2538-bsl/pull/155 was not meant on purpose.

vjuergens, 2024-03-17:
The patch "Fix broken port arg #163" is working for me, unfortunealy I am facing now issue #162 "force_speed"

alesms, 2024-03-17:
any news @JelmerT @git-developer?

git-developer, 2024-03-17:
No need to run out of patience:
* A [workaround](https://github.com/JelmerT/cc2538-bsl/issues/161#issuecomment-2002320386) was advertised in this thread if you need a working image.
* The issue has an open [PR!163](https://github.com/JelmerT/cc2538-bsl/pull/163). 
* In case you're affected by the issue #162, there's an open [PR!165](https://github.com/JelmerT/cc2538-bsl/pull/165), too.

Subscribe the PRs and you will be notified on any change.

accordxtc, 2024-03-17:
I can confirm using the previous version I was able to update with no issues. Thanks for the quick work around.

git-developer, 2024-03-17:
To minimize further irritations, I updated the `latest` tag of image `ckware/ti-cc-tool` to point to the last working version `2024-01-28`.

JelmerT, 2024-03-18:
Fixed with #163 👍 
Thanks for reporting @6wheels

closed 2024-03-18 by JelmerT

git-developer, 2024-03-18:
The container image is updated.

6wheels, 2024-03-18:
Many thanks for this fast fix!