JelmerT/cc2538-bsl #162: ERROR: 'Namespace' object has no attribute 'force_speed'

wesgeorge, 2024-03-15:
Attempting to flash a sonoff 3p using both the standalone python scripts and the docker command, and running into the same error. 

Here's the output of a verbose run: 

sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip ckware/ti-cc-tool -ewv -p 0 --bootloader-sonoff-usb -V
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20230507.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20230507.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Firmware file: Intel Hex
Connecting to target...
*** sending synch sequence
Got 0 additional bytes before ACK/NACK
*** GetChipId command (0x28)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Version 0x3202F000
    Unrecognized chip ID. Trying CC13xx/CC26xx
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:25:8D:D7:40
Traceback (most recent call last):
  File "/usr/local/bin/cc2538-bsl.py", line 1150, in <module>
    if args.force_speed != 1 and device.has_cmd_set_xosc:
       ^^^^^^^^^^^^^^^^
AttributeError: 'Namespace' object has no attribute 'force_speed'
ERROR: 'Namespace' object has no attribute 'force_speed'

git-developer, 2024-03-17:
Looks like another regression caused by https://github.com/JelmerT/cc2538-bsl/pull/155

git-developer, 2024-03-17:
I've built an image with the proposed fix, please try `ckware/ti-cc-tool:2024-03-17` and report back.

git-developer, 2024-03-17:
[According](https://github.com/git-developer/cc2538-bsl/commit/5ddd9c963a49c8dc20dc8b7f1530d3f93c148705#commitcomment-139876220) to @alesms, my proposal this does not heal the issue. This attempt was just a shot in the dark, thus it needs further investigation by @Ayush1325  and/or @JelmerT. I changed my PR to draft status meanwhile.

Ayush1325, 2024-03-17:
Found the problem. The force_speed property was originally set if `-b` was specified and was not a cli argument. Since it does not exist as an argument, `args.force_speed` is always false.  It can be fixed as follows:

```py
# Remove default value
 parser.add_argument('-b', '--baud', type=int, help='Baud speed (default = 500000)')

..

force_speed = False
if args.baud:
    force_speed = True
else:
    args.baud = 500000
```

I apologize for the regressions.

git-developer, 2024-03-18:
The container image is updated.

wesgeorge, 2024-03-18:
Thanks for the update! retested and this appears to be working, as it completed the update, but it still errors out at the end: 

wes@tantive-iv:~$ sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip ckware/ti-cc-tool -ewv --bootloader-sonoff-usb -V
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20230507.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20230507.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Firmware file: Intel Hex
Connecting to target...
*** sending synch sequence
Got 0 additional bytes before ACK/NACK
*** GetChipId command (0x28)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Version 0x3202F000
    Unrecognized chip ID. Trying CC13xx/CC26xx
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
*** Mem Read (0x2A)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
[... 4394 lines cut ...]
*** Send Data (0x24)
Got 0 additional bytes before ACK/NACK
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Write done
Verifying by comparing CRC32 calculations.
*** CRC32 command(0x27)
Got 0 additional bytes before ACK/NACK
*** received 6 bytes
*** GetStatus command (0x23)
Got 0 additional bytes before ACK/NACK
*** received 3 bytes
Command Successful
    Verified (match: 0xe83aa727)
Traceback (most recent call last):
  File "/usr/local/bin/cc2538-bsl.py", line 1225, in <module>
    ieee_addr = parse_ieee_address(args.ieee_address)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/cc2538-bsl.py", line 967, in parse_ieee_address
    return int(inaddr, 16)
           ^^^^^^^^^^^^^^^
TypeError: int() can't convert non-string with explicit base
ERROR: int() can't convert non-string with explicit base

git-developer, 2024-03-19:
Looks like another regression caused by https://github.com/JelmerT/cc2538-bsl/pull/155. Could you please check if `ckware/ti-cc-tool:2024-03-19` works for you?

git-developer, 2024-03-23:
Could you please check if `ckware/ti-cc-tool:2024-03-19` works for you, @wesgeorge?

wesgeorge, 2024-03-25:
Apologies if I was unclear, the firmware update completed successfully, so the error that I posted was some unrelated and I guess cosmetic issue. Closing this issue out.

closed 2024-03-25 by wesgeorge

git-developer, 2024-03-26:
Thanks, @wesgeorge! Nonetheless, it would be nice if you could test whether `ckware/ti-cc-tool:2024-03-19` works for you, as test for for pending fix https://github.com/JelmerT/cc2538-bsl/pull/168.