JelmerT/cc2538-bsl #170: "ERROR: int() can't convert non-string with explicit base" when flashing CC2652P

Ma-Str, 2024-03-27:
I have a problem while flashing my CC2652P. It says it's done verifying with a match but then goes on with an int error.

`sonoff
Opening port COM3, baud 500000
Reading data from .\CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:30:CB:BF:CB
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xe83aa727)
ERROR: int() can't convert non-string with explicit base`

When trying to start zigbee2mqtt it fails

Zigbee2MQTT:info  2024-03-27 17:31:55: Starting Zigbee2MQTT version 1.36.0 (commit #unknown)
Zigbee2MQTT:info  2024-03-27 17:31:55: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:error 2024-03-27 17:32:06: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-27 17:32:06: Failed to start zigbee
Zigbee2MQTT:error 2024-03-27 17:32:06: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-27 17:32:06: Exiting...
Zigbee2MQTT:error 2024-03-27 17:32:06: undefined

Any Idea on how to fix this?

Thanks!

closed 2024-03-27 by Ma-Str

ccle610, 2024-03-29:
I already have exactly the same Problem and I'm very intrested in solving it, since I work hard on it since 4 Hour . . . 
So, con you please tell me, how to solve thois Problem?

ccle610, 2024-03-30:
Fixed in #168.