JelmerT/cc2538-bsl #172: ERROR: int() can't convert non-string with explicit base

McP4nth3r, 2024-04-07:
I'm using the `SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N Zigbee USB Stick,Zigbee Hub for ZHA in Home Assistant or Zigbee2MQTT,Open HAB etc.`
i want to flash the newsest firmware with your python script and became this error:

```python cc2538-bsl.py -p COM6 -evw --bootloader-sonoff-usb .\CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
sonoff
Opening port COM6, baud 500000
Reading data from .\CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:2E:1E:06:48
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xe83aa727)
ERROR: int() can't convert non-string with explicit base```

frober, 2024-04-07:
I've the same issue.

mathiasth, 2024-04-07:
Same issue here as well, with the same stick. Thanks in advance!

bolovanos, 2024-04-10:
Same.
Python 3.11.9  (windows 11 64bit)
SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

`python cc2538-bsl.py -p COM8 -e -v -w --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
sonoff
Opening port COM8, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: xx:xx:xx:xx:xx:xx:xx:xx
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xa9dc145d)
ERROR: int() can't convert non-string with explicit base`

In my case I failed even with  uartLog.py and Flash Programmer 2 described in [flashing instruction](https://github.com/Koenkk/zigbee2mqtt.io/raw/master/docs/guide/adapters/flashing/zigbee-3.0-usb-dongle-plus-firmware_flashing-1-1.pdf ).

Be careful with that  latest (20230507) it has [some issues](https://github.com/Koenkk/Z-Stack-firmware/discussions/474) and [496](https://github.com/Koenkk/Z-Stack-firmware/discussions/496).

frober, 2024-04-10:
Thanks @bolovanos , with the patch for the cc2538-bsl.py it works. 

```
D:\Fhem\Zigbee\cc2538-bsl-master>python cc2538-bsl.py -p COM3 -evw --bootloader-sonoff-usb .\CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
sonoff
Opening port COM3, baud 500000
Reading data from .\CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: ......
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xa9dc145d)
```

AndreasBeyer, 2024-04-17:
I used also the https://github.com/JelmerT/cc2538-bsl/pull/168 patch and worked like a charm on Windows 11 Pro N, Python 3.10.7 and SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N)

mdeweerd, 2024-04-25:
I got the same result, applied `python -m trace -t <REGULAR COMMAND>`, amended the code to find that inaddr is None in the following code (in ValueError):

```python
# Convert the entered IEEE address into an integer
def parse_ieee_address(inaddr):
    try:
        return int(inaddr, 16)
    except ValueError:
        sys.stdout.write(f"ValueError: IEEE address is reported as {inaddr!r}")
# Convert the entered IEEE address into an integer
def parse_ieee_address(inaddr):
    try:
        return int(inaddr, 16)
    except ValueError:
        sys.stdout.write(f"ValueError: IEEE address is reported as {inaddr!r}")
```

I'll try the patch as well.

SKART1, 2024-05-18:
The problem looks like in new behaviour of arg parser library which return None insead of 0 as default

Patch looks like correct for modern python versions.

As for mee I reccomend explicitly write types of arguments model to allow linter to find such cases

H1ghSyst3m, 2024-06-01:
Was the code already patched? I am using Python 3.12.3 and I still got get error.

```
(venv) ➜  cc2538-bsl sudo python3 cc2538-bsl.py -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20221226.hex 
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:2C:44:F7:31
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xa9dc145d)
ERROR: int() can't convert non-string with explicit base
```

afiedler, 2024-06-05:
It hasn't been applied. It's a one-liner though in https://github.com/JelmerT/cc2538-bsl/pull/168. Easy to make a local change to the script. With that, I confirmed it's working with Python 3.12.2.

defanator, 2024-07-22:
Just caught this one with SONOFF ZBDongle-P; fix from https://github.com/JelmerT/cc2538-bsl/pull/168 helped.

alfs, 2024-08-24:
I had the same problem. Google led me here. Patch in #164 resolved the problem, thanks.

esand, 2024-08-25:
Same issue as others doing a standard -ewv. If I'm looking at the code correctly, if I were doing only -ewv and got this error at the end after the CRC check, I shouldn't have to worry, right? It seems everything is working, and I didn't need to change the IEEE address.

srett, 2024-09-07:
Same issue here. Fixed via

```python
if args.ieee_address != 0 and args.ieee_address != None:
```