JelmerT/cc2538-bsl #176: bytearrary index out of range

grod55, 2024-07-04:
Hi, I'm trying to Flash coordinator FW and keep getting "bytearrary index out of range". As you can see below, everything seems to do what it is supposed to do until it starts writing at the end.

Any suggestions to resolve this issue?

Thanks,

pi:~ $ docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip ckware/ti-cc-tool -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb
Unable to find image 'ckware/ti-cc-tool:latest' locally
latest: Pulling from ckware/ti-cc-tool
fda0ff469afd: Pull complete
bd70e72a4529: Pull complete
2620ee775689: Pull complete
b63782bf4aa7: Pull complete
8ca19e7c644b: Pull complete
e0cae168a165: Pull complete
5ce45cc6e57f: Pull complete
40b8d839b7eb: Pull complete
4f4fb700ef54: Pull complete
Digest: sha256:360d92c2978d181cd86829a4fe80bc9f28418f449461aec10dedddc5c9c7b1a1
Status: Downloaded newer image for ckware/ti-cc-tool:latest
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x                .0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20230507.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20230507.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:30:C6:EB:39
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
ERROR: bytearray index out of range

gcs8, 2024-09-02:
Same here

`sonoff_temp # docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 --group-add audio --privileged -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/releases/download/Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_launchpad_coord
inator_20240710.zip ckware/ti-cc-tool -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/releases/download/Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20240710.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20240710.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Firmware file: Intel Hex
Connecting to target...
ERROR: bytearray index out of range
sonoff_temp #
`

verschaevesiebe, 2025-03-05:
For anyone coming back later to this thread.
I got this error due to the usb device being in use. 

If you're running Zigbee2MQTT already through docker try stopping that container and retry the docker command again to do the firmware upgrade. This should work 👍🏽