JelmerT/cc2538-bsl #180: Flashing failed due to CRC32 calculation error

Chris3467893, 2024-09-10:
I have a SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N Zigbee USB Stick integrated with my Home Assistant green. Last week I receive a message in HA, that the Dongle requires a firmware update. I commited this requests. After automatic updating and reboot was finished the Zigbee Dongle reported an error: “Network formation failed: NVRAM is corrupted”.
After several tries to fix the problem with lower impact actions I decided to flash the firmware with CC1352P2_CC2652P_launchpad_coordinator_20240710.hex. MD5Checksum of this file is 0f16bf0f96968e6b0168c2bc0a8eeefd.

The update process was
(venv) ➜ cc2538-bsl sudo python3 cc2538-bsl.py -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb ./CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:2C:44:FD:F5
Performing mass erase
Erasing all main bank flash sectors
Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
Write done
Verifying by comparing CRC32 calculations.
Verified (match: 0xd9dd0124)
ERROR: int() can't convert non-string with explicit base

Is this caused by the firmware? What should I do to repair my dongle?

For this topic I opened an issue in [https://github.com/Koenkk/Z-Stack-firmware/issues/520](https://github.com/Koenkk/Z-Stack-firmware/issues/520) and received the answer that this is an issue in for cc2538-bsl

TheQue42, 2024-09-12:
As described here, this can be manually fixed: 

https://github.com/JelmerT/cc2538-bsl/issues/172

until someone with access merges https://github.com/JelmerT/cc2538-bsl/pull/168

JelmerT, 2024-09-23:
should be fixed no 👍

closed 2024-09-23 by JelmerT